morrolinux/mpradio #1: raspberry pi 3

raffibedo, 2017-05-14:
It isnt working in my raspberry pi 3. i used the install script but bt is rejecting my nexus

raffibedo, 2017-05-14:
i tested with the mediafire image and i cant connect my nexus 5x with the raspberry. any idea?

morrolinux, 2017-05-15:
Hi.
Yep, the bluetooth rejection for some device was a known problem in the mediafire image, which is being solved in the latest version on this repo. 
you maybe downloaded the mediafire image, and then tried the install script from github to fix it, but i'm afraid ATM the copy instructions of the install script form github does not overwrite existing files.
so to fix this you could either:

1) git clone this repo in a temporary directory different form your pi's home (eg: /home/pi/temp/)
2) remove clean by moving to the install folder and run "sudo bash install.sh remove"
3) start over from step 1 but this time, on step 2 run "sudo bash install.sh" (without remove flag)

or
1) fresh/new install raspbian from a clean raspbian image
2) git clone this repo in a directory different form /home/pi (as above)
3) run the install script

please let me know if you had further questions/troubles

closed 2017-05-17 by morrolinux