morrolinux/mpradio #12: raspberry pi zero w

ljcp, 2017-09-20:
Hi there great project,

I have a fresh install of 

RASPBIAN STRETCH LITE
Minimal image based on Debian Stretch
Version:September 2017
Release date:2017-09-07
Kernel version:4.9
Release notes:Link

I run the install/install.sh

gcc -Wall -std=gnu99 -c -g -O3 -march=armv6 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=1 control_pipe.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv6 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=1 mailbox.c
gcc -o pi_fm_rds rds.o waveforms.o mailbox.o pi_fm_rds.o fm_mpx.o control_pipe.o -lm -lsndfile
Installation completed! Rebooting in 10 seconds...

it reboots but i'm not seeing any device to pair in my android Bluetooth menu , any insights on to troubleshoot it ?

thank you in advance

ljcp, 2017-09-20:
well , i enter to bluetoothctl and i did

power on 
scan on 
pair android:address

and now is connected but every time i restart the pi i had to  power on and connect to my android device

and so far the fm is not working , the systemctl status mpradio , looks fine , anyother logs to check ? thank you

morrolinux, 2017-09-21:
Hello
As for the 1st question: 
Could you try if manually issuing the following commands is making your Pi visible and pairable?
	/usr/bin/sudo -u pi -s pulseaudio -D
	hciconfig hci0 piscan
if that so, i might think the play_storage() function isn't being invoked, and therefore init() neither. 
In that case, you could:
1) connect a usb key containing at least one mp3 track 
2) patch the code for invoking init() somewhere else, like in the main function.
but for now, just let me know how it goes with those commands.
 
As for the 2nd question:
by default, the pi should be broadcasting on FM 88.8 and you should plug a wire on the 4th pin to function as antenna (like so http://www.skagmo.com/temp/content/projects/22_pihat/img/gpio_1200.jpg )

ljcp, 2017-09-21:
thank you for the response , going to try it !

ljcp, 2017-09-22:
I look at your code more closely and running

1. ./simple-agent
2. bluetoothctl (as root user ) -> power on , connect 
3. checking pactl list sources short ( with pi user )
4. /bin/su pi -c "parec -d **frompactl list**  | sox -t raw -v 1.3 -G -b 16 -e signed -c 2 -r 44100 - -t wav - | sudo /home/pi/PiFmRds/src/pi_fm_rds -ps 'BLUETOOTH' -rt 'A2DP BLUETOOTH' -freq 103.8 -audio -"

it worked just fine 👍  , if I'm using Spotify and I pause it too long or too fast, the process hangs and needs to be re-run again the command , not sure if you handle that case in your C code , 

./bluetooth neverworked for me
mpradio neither so i just grabbed the command

so i'm thinking on doing a new python

1. turns on bluetooth
2. runs ./simple-agent ( with some fixes )
3. when a bluetooth device is connected runs the parec... command 

thank you

morrolinux, 2017-09-22:
well, step 2 is interactive that's why we wanted to avoid it... anyways, keep me updated if you come up with something interesting 
bye

morrolinux, 2017-09-29:
i think i found what was wrong:
- simple-agent was not being launched correctly from crontab. (i created a systemd unit to fix that)
- a recent pulseaudio update changed the convention for a2dp device names. (fixed)
you are encouraged to try it if still interested.

closed 2017-10-11 by morrolinux