morrolinux/mpradio #13: Does not appear to work on RPI3

PedroC1999, 2017-10-11:
Hello!

Just saw your write up on your blog and I would like to try this myself, I ran the install instructions as per your Github and the RPI appears to be broadcasting sits bluetooth and I can succesfully connect. However the Pi does not appear to be transmitting and I have no idea how to fetch a log or debug this. Could you help guide me?

morrolinux, 2017-10-11:
Hello, which frequency have you put in pirateradio.config? default freq. is 88.8 if you don't set one.
Also, Have you tried attaching a usb key with at least one mp3 in it? I haven't tested it yet, but i think the raspberry could loop (restarting) forever if no mp3 to play is found, making it impossible to play even bluetooth. 
Thanks for your feedback, please let me know if it helps :)

PedroC1999, 2017-10-11:
I've set it at 108.0 to make it easier to find in my radio (no digital tuner) 

I haven't tried it, is it a case of place the MP3 in the root of the USB? 

Also, is there a way to even check its running? I just reboot my Pi and its "working" or do I need to launch any script?

morrolinux, 2017-10-11:
You can put a mp3 file in any directory of your USB key, even the root or a
sub-folder. Of course you can start it, stop it, and even check if it's
running by using the standard systemd commands since it's a service:

sudo systemctl status mpradio
sudo systemctl start mpradio
sudo systemctl stop mpradio
Etc...
I'm going to add some more instructions in the readme of the project so
that everyone can read them directly in the home page

PedroC1999, 2017-10-11:
Okay,

I get home relatively soon and will be able to test this for you. I'll happily work with you in testing if you require anything

Thanks again, Pedro

PedroC1999, 2017-10-11:
Below is the output of the status command, does not appear to be transmitting the MP3 either. Clean install on Rasbian

`pi@raspberrypi:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: 
   Active: active (running) since Wed 2017-10-11 17:15:03 UTC; 3h 0min ago
 Main PID: 474 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─  474 /home/pi/mpradio
           ├─  696 pulseaudio -D
           ├─13455 sh -c systemctl force-reload udev systemd-udevd-control.socke
           └─13456 systemctl force-reload udev systemd-udevd-control.socket syst

Oct 11 20:15:47 raspberrypi pulseaudio[13421]: pa_pid_file_create() failed.
Oct 11 20:15:47 raspberrypi bash[474]: E: [pulseaudio] main.c: Daemon startup fa
Oct 11 20:15:47 raspberrypi sudo[13414]: pam_unix(sudo:session): session closed 
Oct 11 20:15:47 raspberrypi bash[474]: find: ‘/pirateradio’: No such file or dir
Oct 11 20:15:47 raspberrypi sudo[13439]:     root : TTY=unknown ; PWD=/ ; USER=p
Oct 11 20:15:47 raspberrypi sudo[13439]: pam_unix(sudo:session): session opened 
Oct 11 20:15:47 raspberrypi pulseaudio[13445]: Daemon already running.
Oct 11 20:15:47 raspberrypi pulseaudio[13445]: pa_pid_file_create() failed.
Oct 11 20:15:47 raspberrypi bash[474]: E: [pulseaudio] main.c: Daemon startup fa
Oct 11 20:15:47 raspberrypi sudo[13439]: pam_unix(sudo:session): session closed 
lines 1-20/20 (END)...skipping...
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-10-11 17:15:03 UTC; 3h 0min ago
 Main PID: 474 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─  474 /home/pi/mpradio
           ├─  696 pulseaudio -D
           ├─13455 sh -c systemctl force-reload udev systemd-udevd-control.socket systemd-udevd-kernel.socket
           └─13456 systemctl force-reload udev systemd-udevd-control.socket systemd-udevd-kernel.socket

Oct 11 20:15:47 raspberrypi pulseaudio[13421]: pa_pid_file_create() failed.
Oct 11 20:15:47 raspberrypi bash[474]: E: [pulseaudio] main.c: Daemon startup failed.
Oct 11 20:15:47 raspberrypi sudo[13414]: pam_unix(sudo:session): session closed for user pi
Oct 11 20:15:47 raspberrypi bash[474]: find: ‘/pirateradio’: No such file or directory
Oct 11 20:15:47 raspberrypi sudo[13439]:     root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c pulseaudio -D
Oct 11 20:15:47 raspberrypi sudo[13439]: pam_unix(sudo:session): session opened for user pi by (uid=0)
Oct 11 20:15:47 raspberrypi pulseaudio[13445]: Daemon already running.
Oct 11 20:15:47 raspberrypi pulseaudio[13445]: pa_pid_file_create() failed.
Oct 11 20:15:47 raspberrypi bash[474]: E: [pulseaudio] main.c: Daemon startup failed.
Oct 11 20:15:47 raspberrypi sudo[13439]: pam_unix(sudo:session): session closed for user pi
~
`

morrolinux, 2017-10-11:
Thanks for the trial. Have you already tried after booting with a usb key loaded with one mp3 file?

PedroC1999, 2017-10-11:
`
pi@raspberrypi:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset:
   Active: active (running) since Wed 2017-10-11 20:36:23 UTC; 20min ago
 Main PID: 410 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─  410 /home/pi/mpradio
           ├─  645 pulseaudio -D
           ├─10269 sh -c /usr/bin/sudo -u pi -s pulseaudio -D
           ├─10271 /usr/bin/sudo -u pi -s pulseaudio -D
           └─10276 pulseaudio -D

Oct 11 20:56:31 raspberrypi pulseaudio[10224]: pa_pid_file_create() failed.
Oct 11 20:56:31 raspberrypi bash[410]: E: [pulseaudio] main.c: Daemon startup fa
Oct 11 20:56:31 raspberrypi sudo[10218]: pam_unix(sudo:session): session closed
Oct 11 20:56:31 raspberrypi sudo[10242]:     root : TTY=unknown ; PWD=/ ; USER=p
Oct 11 20:56:31 raspberrypi sudo[10242]: pam_unix(sudo:session): session opened
Oct 11 20:56:31 raspberrypi pulseaudio[10248]: Daemon already running.
Oct 11 20:56:31 raspberrypi bash[410]: E: [pulseaudio] main.c: Daemon startup fa
Oct 11 20:56:31 raspberrypi sudo[10242]: pam_unix(sudo:session): session closed
Oct 11 20:56:31 raspberrypi sudo[10271]:     root : TTY=unknown ; PWD=/ ; USER=p
Oct 11 20:56:32 raspberrypi sudo[10271]: pam_unix(sudo:session): session opened
lines 1-21/21 (END)...skipping...
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-10-11 20:36:23 UTC; 20min ago
 Main PID: 410 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─  410 /home/pi/mpradio
           ├─  645 pulseaudio -D
           ├─10269 sh -c /usr/bin/sudo -u pi -s pulseaudio -D
           ├─10271 /usr/bin/sudo -u pi -s pulseaudio -D
           └─10276 pulseaudio -D

Oct 11 20:56:31 raspberrypi pulseaudio[10224]: pa_pid_file_create() failed.
Oct 11 20:56:31 raspberrypi bash[410]: E: [pulseaudio] main.c: Daemon startup failed.
Oct 11 20:56:31 raspberrypi sudo[10218]: pam_unix(sudo:session): session closed for user pi
Oct 11 20:56:31 raspberrypi sudo[10242]:     root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c p
Oct 11 20:56:31 raspberrypi sudo[10242]: pam_unix(sudo:session): session opened for user pi by (uid=0)
Oct 11 20:56:31 raspberrypi pulseaudio[10248]: Daemon already running.
Oct 11 20:56:31 raspberrypi bash[410]: E: [pulseaudio] main.c: Daemon startup failed.
Oct 11 20:56:31 raspberrypi sudo[10242]: pam_unix(sudo:session): session closed for user pi
Oct 11 20:56:31 raspberrypi sudo[10271]:     root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c p
Oct 11 20:56:32 raspberrypi sudo[10271]: pam_unix(sudo:session): session opened for user pi by (uid=0)
`

PedroC1999, 2017-10-11:
heres the output now, one mp3 onboard

morrolinux, 2017-10-11:
seems like it's running.. do you hear anything?

PedroC1999, 2017-10-11:
I will have to set it up in my car, I don't think this radio is picking it up. I am using about 25cm of single core unshielded bare wire and I am within 50cm of the radio.

Does that sound alright?

morrolinux, 2017-10-12:
Yea it should be okay! i wonder why you weren't able to pick it up... well
let me know how it goes when you get in your car (y)

PedroC1999, 2017-10-12:
Nope! Couldn't get it to work. 1 file on the USB indicating it to broadcast on 108.0, and another being the MP3 file. 

Also tried connecting to blue tooth and still nothing would play

PedroC1999, 2017-10-12:
Would it be possible to have a recent DD image send to me or uploaded?

morrolinux, 2017-10-12:
Well, that wouldn't work out of the box, since PiFmRDS needs to be
recompiled upon hardware change... BUT.. if you would help me debugging
this, i have few extra questions to ask you and everyone else would benefit
from it too...

can you please type $ ps aux | grep -i pifm
and paste here the output? there should be a command line with all
parameters used to launch the FM streamer application, so we can check if
it's running on the right settings

PedroC1999, 2017-10-12:
pi@raspberrypi:~ $ ps aux | grep -i pifm
pi       29774  0.0  0.0   4372   568 pts/1    S+   15:58   0:00 gr
p --color=auto -i pifm
pi@raspberrypi:~ $

Here you go - what do you think of that?

morrolinux, 2017-10-12:
Alright, just try it now please, i think i overlooked something in settingsprovider.cc

PedroC1999, 2017-10-12:
`
pi@raspberrypi:~ $ sudo systemctl status mpradio                                                                                                                                                                                             ● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset:
   Active: active (running) since Thu 2017-10-12 18:22:57 UTC; 7min ago
 Main PID: 470 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─ 470 /home/pi/mpradio
           ├─ 638 pulseaudio -D
           ├─4752 sh -c /usr/bin/sudo -u pi -s pulseaudio -D
           ├─4753 /usr/bin/sudo -u pi -s pulseaudio -D
           └─4758 pulseaudio -D

Oct 12 18:30:22 raspberrypi bash[470]: E: [pulseaudio] main.c: Daemon startup fa
Oct 12 18:30:22 raspberrypi sudo[4705]: pam_unix(sudo:session): session closed f
Oct 12 18:30:22 raspberrypi sudo[4729]:     root : TTY=unknown ; PWD=/ ; USER=pi
Oct 12 18:30:22 raspberrypi sudo[4729]: pam_unix(sudo:session): session opened f
Oct 12 18:30:22 raspberrypi pulseaudio[4735]: Daemon already running.
Oct 12 18:30:22 raspberrypi pulseaudio[4735]: pa_pid_file_create() failed.
Oct 12 18:30:22 raspberrypi bash[470]: E: [pulseaudio] main.c: Daemon startup fa
Oct 12 18:30:22 raspberrypi sudo[4729]: pam_unix(sudo:session): session closed f
Oct 12 18:30:23 raspberrypi sudo[4753]:     root : TTY=unknown ; PWD=/ ; USER=pi
Oct 12 18:30:23 raspberrypi sudo[4753]: pam_unix(sudo:session): session opened f
lines 1-21/21 (END)...skipping...
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-10-12 18:22:57 UTC; 7min ago
 Main PID: 470 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─ 470 /home/pi/mpradio
           ├─ 638 pulseaudio -D
           ├─4752 sh -c /usr/bin/sudo -u pi -s pulseaudio -D
           ├─4753 /usr/bin/sudo -u pi -s pulseaudio -D
           └─4758 pulseaudio -D

Oct 12 18:30:22 raspberrypi bash[470]: E: [pulseaudio] main.c: Daemon startup failed.
Oct 12 18:30:22 raspberrypi sudo[4705]: pam_unix(sudo:session): session closed for user pi
Oct 12 18:30:22 raspberrypi sudo[4729]:     root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c pulseaudio -D
Oct 12 18:30:22 raspberrypi sudo[4729]: pam_unix(sudo:session): session opened for user pi by (uid=0)
Oct 12 18:30:22 raspberrypi pulseaudio[4735]: Daemon already running.
Oct 12 18:30:22 raspberrypi pulseaudio[4735]: pa_pid_file_create() failed.
Oct 12 18:30:22 raspberrypi bash[470]: E: [pulseaudio] main.c: Daemon startup failed.
Oct 12 18:30:22 raspberrypi sudo[4729]: pam_unix(sudo:session): session closed for user pi
Oct 12 18:30:23 raspberrypi sudo[4753]:     root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c pulseaudio -D
Oct 12 18:30:23 raspberrypi sudo[4753]: pam_unix(sudo:session): session opened for user pi by (uid=0)
`

`pi@raspberrypi:~ $ ps aux | grep -i pifm
pi        9021  0.0  0.0   4372   536 pts/0    S+   18:31   0:00 grep --color=auto -i pifm
pi@raspberrypi:~ $
pi@raspberrypi:~ $ sudo systemctl status mpradio                                                                                                                                                                                             ● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset:
   Active: active (running) since Thu 2017-10-12 18:22:57 UTC; 7min ago
 Main PID: 470 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─ 470 /home/pi/mpradio
           ├─ 638 pulseaudio -D
           ├─4752 sh -c /usr/bin/sudo -u pi -s pulseaudio -D
           ├─4753 /usr/bin/sudo -u pi -s pulseaudio -D
           └─4758 pulseaudio -D

Oct 12 18:30:22 raspberrypi bash[470]: E: [pulseaudio] main.c: Daemon startup fa
Oct 12 18:30:22 raspberrypi sudo[4705]: pam_unix(sudo:session): session closed f
Oct 12 18:30:22 raspberrypi sudo[4729]:     root : TTY=unknown ; PWD=/ ; USER=pi
Oct 12 18:30:22 raspberrypi sudo[4729]: pam_unix(sudo:session): session opened f
`

PedroC1999, 2017-10-12:
cant hear anything still, do you require anything else?

morrolinux, 2017-10-12:
Maybe I forgot to mention I just updated the repository, have you installed
the latest version already? Also, please try to plug a USB key with one mp3
on it (no strange characters in the name) and no configuration file.
Reboot, Then you should really try to tune 88.8 on your radio and see what
happens.

PedroC1999, 2017-10-12:
I will update again, I had done it once but shall do it again just incase

PedroC1999, 2017-10-12:
i also renamed the mp3 to song.mp3

morrolinux, 2017-10-12:
Damn, I'm not sure since when crudini is case sensitive... anyway I just noticed i was not able to set a custom frequency because of it:
As I'm writing, I just updated the settingsprovider.cc to match [PIRATERADIO] instead of [pirateradio] (uppercase instead of lowercase) which is more ini-compliant. 
I'm sorry for those continuous updates, but I think we're going to make it.

PedroC1999, 2017-10-12:
Dont be sorry, just hope to get this working!

PiFM has worked on my Pi before, and I have reset it multiple times since.

Again, I did not hear anything. Does PiFM produce any logs etc?

morrolinux, 2017-10-12:
This could be not just about PiFm but more likely mpradio (my implementation) which has few weaknesses. Anyway, you can stop mpradio service and launch /home/pi/mpradio manually from shell. You should then see something useful on the screen

morrolinux, 2017-10-12:
anyway you should fetch and update to the latest commit: i personally tested and seems to be working alright with a usb key:
- song.mp3
- pirateradio.config:
[PIRATERADIO]
frequency=107.0

and even without the config file, it's correctly defaulting to 88.8..
i also solved a process stall that used to keep the cpu 100% busy restarting forever the process, when no mp3 was provided. 
That said, it's all to be tested now :)

PedroC1999, 2017-10-12:
I have removed the config file anyway to remove more variables. Trying now

PedroC1999, 2017-10-12:
just to add, I am using the RPi 3's built in bluetooth and connecting/pairing works fine

morrolinux, 2017-10-12:
If you are picking up 107.0 , that means your USB key is not mounted, and
mpradio is picking up the default configuration file located at
/pirateradio/pirateradio.config (which is masked with the real usb key
root, when a usb key is plugged) therefore, there is no mp3 file to play
and the program quits without playing anything.

As for the bluetooth message: if you launch ./mpradio without parameters,
it will always give you that message. there is a specific udev script in
charge of launching mpradio with the bluez_device.YOUR_BT_ADDRESS.bt_audio
(or something) automatically when you connect via bluetooth... but we will
take care about it later on..

for the moment, you should try to understand why the usb key is not mounted
at all, and remember:
1) USB key should be in FAT32 filesystem
2) USB key should have just one partition
3) msdos partition schema, NOT GPT

$ mount | grep pirateradio
should give you something like /dev/sda1   /pirateradio/
if the key is mounted, or nothing if is not.

also try $ ls /pirateradio/ to see if your mp3 is there (which is not, ATM)

PedroC1999, 2017-10-12:
I have a USB2.0 device plugged into the Pi3, its formatted as "DATABAR" and its accessible from the Pi's file explorer, where the song is visible.

PedroC1999, 2017-10-12:
The MP3 is not listed when I run that command, Pirateradio.config is (this is not on my USB drive)

morrolinux, 2017-10-12:
1) I think the USB key is mounted in a different location than
/pirateradio/ hence, you see pirateradio.conf and not your mp3 (which is
not the content of your USB key)
2) you should really disable graphical sessions on your raspberry or at
least kill them, as they can intercept new devices upon connection and
cause the problem in point 1)

PedroC1999, 2017-10-12:
I am not great at Unix commands and do struggle slightly - sometimes I find it useful. 

Would it be possible to run a command to mount it properly

morrolinux, 2017-10-12:
You have to unmount it from graphical session first, and then run $ sudo
mount -a to make it happen.
That said, graphical session is nothing useful for what we're doing here,
and can cause so much trouble.. consider disabling it before you put your
Pi into your car, to avoid having to type those commands before driving ;)

PedroC1999, 2017-10-12:
Oh yes definitely haha, 

I unmounted the volume in GUI and disconnected. Then I ran the mount command and the outputs were the same

morrolinux, 2017-10-13:
Ok, let's try to start over:
1) disconnect your graphical session (disable it, even temporarily, would be optimal)
2) $ sudo umount /pirateradio (just to make sure)
3) ensure your usb is plugged and $ sudo mount /dev/sda1 /pirateradio
4) $ ls /pirateradio   should list your mp3 on the on the USB key
5) $ sudo systemctl stop mpradio && sudo systemctl start mpradio
6) now you should start hearing something on 88.8 (if there's no config file)

PedroC1999, 2017-10-13:
pi@raspberrypi:~ $ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 7.4 GiB, 7948206080 bytes, 15523840 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x000018b6

Device         Boot   Start      End  Sectors  Size Id Type
/dev/mmcblk0p1         2048  2490234  2488187  1.2G  e W95 FAT16 (LBA)
/dev/mmcblk0p2      2490235 15523839 13033605  6.2G  5 Extended
/dev/mmcblk0p5      2490368  2555901    65534   32M 83 Linux
/dev/mmcblk0p6      2555904  2697215   141312   69M  c W95 FAT32 (LBA)
/dev/mmcblk0p7      2703360 15523839 12820480  6.1G 83 Linux

Disk /dev/sda: 1.9 GiB, 2003828736 bytes, 3913728 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x6f20736b

Device     Boot      Start        End    Sectors   Size Id Type
/dev/sda1        778135908 1919645538 1141509631 544.3G 72 unknown
/dev/sda2        168689522 2104717761 1936028240 923.2G 65 Novell Netware 386
/dev/sda3       1869881465 3805909656 1936028192 923.2G 79 unknown
/dev/sda4       2885681152 2885736650      55499  27.1M  d unknown

Partition table entries are not in disk order.
pi@raspberrypi:~ $ sudo mount /dev/sda /pirateradio
pi@raspberrypi:~ $ ls /pirateradio
pirateradio.config
pi@raspberrypi:~ $ sudo ls /pirateradio
pirateradio.config

PedroC1999, 2017-10-13:
Above is my disk info, as we see we have a 2gb USB device that is connected and supposedly mounted, but is not listed

morrolinux, 2017-10-13:
You don't just mount /dev/sda which is the physical device. You must refer
to the partition to be mounted, and I see you have 4:
sda1
sda2
sda3
sda4
So you have to specify either /dev/sda1 or /dev/sda2 and so on... By
default, mpradio would mount /dev/sda1 but maybe in your USB key that's not
a fat32 partition. To make things easier I would suggest you to format your
USB key on your computer making just one big FAT partition in which you
place your mp3s

PedroC1999, 2017-10-13:
pi@raspberrypi:~ $ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 7.4 GiB, 7948206080 bytes, 15523840 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x000018b6

Device         Boot   Start      End  Sectors  Size Id Type
/dev/mmcblk0p1         2048  2490234  2488187  1.2G  e W95 FAT16 (LBA)
/dev/mmcblk0p2      2490235 15523839 13033605  6.2G  5 Extended
/dev/mmcblk0p5      2490368  2555901    65534   32M 83 Linux
/dev/mmcblk0p6      2555904  2697215   141312   69M  c W95 FAT32 (LBA)
/dev/mmcblk0p7      2703360 15523839 12820480  6.1G 83 Linux

Disk /dev/sda: 1.9 GiB, 2003828736 bytes, 3913728 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x500a0dff

Device     Boot      Start        End    Sectors   Size Id Type
/dev/sda1       1948285285 3650263507 1701978223 811.6G 6e unknown
/dev/sda2                0          0          0     0B 74 unknown
/dev/sda4         28049408   28049848        441 220.5K  0 Empty

Partition table entries are not in disk order.
pi@raspberrypi:~ $ sudo mount /dev/sda1 /pirateradio
mount: special device /dev/sda1 does not exist
pi@raspberrypi:~ $

PedroC1999, 2017-10-13:
I have reset the drive and it sees the sda1 director but cannot mount it?

morrolinux, 2017-10-13:
On which system are you formatting it? There seem to be still 4 partitions,
and the partition size are totally worng. Can you try another USB drive as
well?

PedroC1999, 2017-10-13:
pi@raspberrypi:~ $ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
[... 63 lines cut ...]
Oct 13 20:31:20 raspberrypi sudo[1082]: pam_unix(sudo:session): session closed f
Oct 13 20:31:20 raspberrypi bash[449]: sox WARN formats: can't set sample rate 4
lines 1-21/21 (END)...skipping...
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2017-10-13 20:17:11 UTC; 16min ago
 Main PID: 449 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─ 449 /home/pi/mpradio
           ├─1088 pulseaudio -D
           ├─1106 sh -c sox -t mp3 -v 1.3 -r 48000 -G "/pirateradio/song.mp3" -t wav - | /home/pi/PiFmRds/src/pi_fm_rds -ctl /home/pi/rds_ctl -ps "song.mp3" -rt "song.mp3" -audio - -freq 88.8
           ├─1107 sox -t mp3 -v 1.3 -r 48000 -G /pirateradio/song.mp3 -t wav -
           └─1108 /home/pi/PiFmRds/src/pi_fm_rds -ctl /home/pi/rds_ctl -ps song.mp3 -rt song.mp3 -audio - -freq 88.8

Oct 13 20:31:19 raspberrypi sudo[1073]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/killall pulseaudio
Oct 13 20:31:19 raspberrypi sudo[1073]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 13 20:31:19 raspberrypi sudo[1073]: pam_unix(sudo:session): session closed for user root
Oct 13 20:31:19 raspberrypi sudo[1082]:     root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c pulseaudio -D
Oct 13 20:31:19 raspberrypi sudo[1082]: pam_unix(sudo:session): session opened for user pi by (uid=0)
Oct 13 20:31:20 raspberrypi pulseaudio[1088]: Default and alternate sample rates are the same.
Oct 13 20:31:20 raspberrypi pulseaudio[1088]: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Oct 13 20:31:20 raspberrypi pulseaudio[1088]: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Oct 13 20:31:20 raspberrypi sudo[1082]: pam_unix(sudo:session): session closed for user pi
Oct 13 20:31:20 raspberrypi bash[449]: sox WARN formats: can't set sample rate 48000; using 44100
~

PedroC1999, 2017-10-13:
here we go, seems to have been the USB drive, going to go test downstairs now!

PedroC1999, 2017-10-13:
I SWEAR i heared the slightest of glimpses of the song, however there was too much interference 

is there anyway to edit the config file on the USB, once its mounted etc and running?

morrolinux, 2017-10-14:
Oh that's great!! We finally reached a good point!
Unfortunately it's not possible to change settings on the fly, but that
should be not big issue:
you will have to tune your radio to seek a station with the lowest possible
signal quality, write it down, and set that frequency in your config file.
The worse is the radio station signal, The better reception you will gain
(less interference with the radio station)
Another trick that might be helpful: i found out that unscrewing the
antenna from your car will help *a lot* improving the reception from your
Pi.

PedroC1999, 2017-10-14:
Hello.

I have been testing on 107.0, there are no radio stations there, just static for +- 1.2MHz

On powerup, I hear the song being played for 0.5 seconds, its obvious that its that. then it stops and I hear nothing, no amount of tuning etc will make me hear it again.

What should my steps be from here?

PedroC1999, 2017-10-14:
pi@raspberrypi:~ $ ls /pirateradio
pirateradio.config  song.mp3  System Volume Information
pi@raspberrypi:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset:
   Active: active (running) since Sat 2017-10-14 00:42:30 UTC; 17h ago
 Main PID: 424 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─424 /home/pi/mpradio
           ├─639 pulseaudio -D
           ├─711 sh -c sox -t mp3 -v 1.3 -r 48000 -G "/pirateradio/song.mp3" -t
           ├─712 sox -t mp3 -v 1.3 -r 48000 -G /pirateradio/song.mp3 -t wav -
           └─713 /home/pi/PiFmRds/src/pi_fm_rds -ctl /home/pi/rds_ctl -ps song.m

Oct 14 00:42:34 raspberrypi sudo[593]: pam_unix(sudo:session): session opened fo
Oct 14 00:42:34 raspberrypi bash[424]: pulseaudio: no process found
Oct 14 00:42:34 raspberrypi sudo[593]: pam_unix(sudo:session): session closed fo
Oct 14 00:42:34 raspberrypi sudo[609]:     root : TTY=unknown ; PWD=/ ; USER=pi
Oct 14 00:42:34 raspberrypi sudo[609]: pam_unix(sudo:session): session opened fo
Oct 14 00:42:35 raspberrypi pulseaudio[639]: Default and alternate sample rates
Oct 14 00:42:35 raspberrypi pulseaudio[639]: Unable to contact D-Bus: org.freede
Oct 14 00:42:35 raspberrypi pulseaudio[639]: Unable to contact D-Bus: org.freede
Oct 14 00:42:35 raspberrypi sudo[609]: pam_unix(sudo:session): session closed fo
Oct 14 00:42:36 raspberrypi bash[424]: sox WARN formats: can't set sample rate 4
lines 1-21/21 (END)
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2017-10-14 00:42:30 UTC; 17h ago
 Main PID: 424 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─424 /home/pi/mpradio
           ├─639 pulseaudio -D
           ├─711 sh -c sox -t mp3 -v 1.3 -r 48000 -G "/pirateradio/song.mp3" -t wav - | /home/pi/PiFmRds/src/
           ├─712 sox -t mp3 -v 1.3 -r 48000 -G /pirateradio/song.mp3 -t wav -
           └─713 /home/pi/PiFmRds/src/pi_fm_rds -ctl /home/pi/rds_ctl -ps song.mp3 -rt song.mp3 -audio - -fre

Oct 14 00:42:34 raspberrypi sudo[593]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 14 00:42:34 raspberrypi bash[424]: pulseaudio: no process found
Oct 14 00:42:34 raspberrypi sudo[593]: pam_unix(sudo:session): session closed for user root
Oct 14 00:42:34 raspberrypi sudo[609]:     root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c pulsea
Oct 14 00:42:34 raspberrypi sudo[609]: pam_unix(sudo:session): session opened for user pi by (uid=0)
Oct 14 00:42:35 raspberrypi pulseaudio[639]: Default and alternate sample rates are the same.
Oct 14 00:42:35 raspberrypi pulseaudio[639]: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported
Oct 14 00:42:35 raspberrypi pulseaudio[639]: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported
Oct 14 00:42:35 raspberrypi sudo[609]: pam_unix(sudo:session): session closed for user pi
Oct 14 00:42:36 raspberrypi bash[424]: sox WARN formats: can't set sample rate 48000; using 44100
~
~
lines 1-21/21 (END)

morrolinux, 2017-10-14:
Strange! Seems allright to me...
Try putting some other songs on the USB drive, it might be some problem
with the mp3 of your choice.

morrolinux, 2017-10-20:
Hello, is everything alright? did you solve?

PedroC1999, 2017-10-28:
Sorry I've been away on Holidays and away from the Pi and a stable internet connection. I haven't yet resolved but also haven't been trying. I'll get back to it soon

start-the-fire, 2018-01-21:
Hi moreno,

your tips regarding the filesystem of the usb stick was helping me out ;)
I was able to get the whole system up & running with my old ipod touch (2nd gen) yesterday but after some time it stoped working and I wasn't able to get it up again. Playing music from the usb stick is still working but when I try to cennct via bluetooth it fails or when it connects it won't play music :/
It seems there is a problem with the bluetooth interface.. I tried it with an installation of debian jessie and stretch. I read in another issue today that you said there is an change according the bluez code

`pi@raspberrypi:/ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled)
   Active: active (running) since Sun 2018-01-21 13:58:23 CET; 5min ago
 Main PID: 727 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─727 /home/pi/mpradio
           ├─783 pulseaudio -D
           ├─811 sh -c sox -t mp3 -v 1.3 -r 48000 -G "/pirateradio/02 Pastel.mp3" -t wav - | /home/pi/PiFmRds/src/pi_fm_rds -ctl /home/pi/rds_ctl -ps "02 Pastel.mp3" -rt "02 Pastel.mp3" -audio - -freq 107.0
           ├─812 sox -t mp3 -v 1.3 -r 48000 -G /pirateradio/02 Pastel.mp3 -t wav -
           └─813 /home/pi/PiFmRds/src/pi_fm_rds -ctl /home/pi/rds_ctl -ps 02 Pastel.mp3 -rt 02 Pastel.mp3 -audio - -freq 107.0

Jan 21 13:58:24 raspberrypi bash[727]: /pirateradio
Jan 21 13:58:24 raspberrypi bash[727]: no bluetooth device provided, playing mp3s...
Jan 21 13:58:24 raspberrypi sudo[776]: root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c pulseaudio -D
Jan 21 13:58:24 raspberrypi sudo[776]: pam_unix(sudo:session): session opened for user pi by (uid=0)
Jan 21 13:58:26 raspberrypi pulseaudio[783]: Default and alternate sample rates are the same.
Jan 21 13:58:26 raspberrypi pulseaudio[783]: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jan 21 13:58:26 raspberrypi pulseaudio[783]: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jan 21 13:58:26 raspberrypi pulseaudio[783]: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Manager" doesn't exist
Jan 21 13:58:26 raspberrypi sudo[776]: pam_unix(sudo:session): session closed for user pi
Jan 21 13:58:26 raspberrypi bash[727]: sox WARN formats: can't set sample rate 48000; using 44100`

Thank for this kick-ass project!
Ben

morrolinux, 2018-01-22:
I just updated the code. check it out :)

start-the-fire, 2018-01-22:
Thank for the very fast respone. I gonna check it out tomorrow morning :)

morrolinux, 2018-01-24:
Fixed. 
Please reopen if you still have trouble related to this issue

closed 2018-01-24 by morrolinux