morrolinux/mpradio #14: Doesn't appear to work on Raspberry Pi Zero W

enduity, 2017-10-13:
I first installed the DD image, but that didn't work at all. So I uninstalled mpradio and installed it from this repo. And MP3 works. But Bluetooth doesn't. I had to install some Bluetooth stuff (notably pi-bluetooth) for Bluetooth to work at all and then make my own simple-agent using a python script I found that replicates it (there was no simple-agent.sh). But it still doesn't work. I get an error, after running mpradio:

```
pi@mpradio:~ $ ./mpradio 40:40:A7:CD:74:46

106.0
/pirateradio
40:40:A7:CD:74:46
Password:
sox WARN wav: Length in output .wav header will be wrong si
ce can't seek to fix it
Using mbox device /dev/vcio.
Allocating physical memory: size = 3403776     mem_ref = 4
   bus_addr = 5f2f6000     virt_addr = 0xb6865000
ppm corr is 0.0000, divider is 1096.4912 (1096 + 2012*2^-12
 [nominal 1096.4912].
Stream error: Invalid argument
Using stdin for audio input.
Input: 44100 Hz, upsampling factor: 5.17
2 channels, generating stereo multiplex.
Created low-pass FIR filter for audio channels, with cutoff
at 12000.0 Hz
PI: 1234, PS: "BLUETOOTH".
RT: "A2DP BLUETOOTH"
Starting to transmit on 106.0 MHz.
Could not rewind in audio file, terminating
Terminating: cleanly deactivated the DMA engine and killed
he carrier.
pi@mpradio:~ $
```

morrolinux, 2017-10-13:
Well,
1) All the needed software to run mpradio is installed as dependency during the install via the installation script install.sh (which you have to run with sudo for it to work proprely)
2) simple-agent too, is installed via the installer (in a tweaked version to allow every connection) so you don't need to install anything else
3) simple-agent is up and running as a systemd service (read the info in the main page of this repo)
4) you don't launch ./mpradio manually except for debugging (because of user swapping issues)
5) it is true that the bluetooth device has to be passed via parameter, but pulseaudio uses another convention for the name of the bluetooth audio sources, therefore, simply indicating the bluetooth address won't work (you can read pulseaudio documentation or bluetooth script in the install folder to see how it's implemented)

6) last but not least, I updated the repo yesterday night, i suggest you to clean what you did before (even a reinstall of raspbian minimal is fine) and install fresh clean. That should work out of the box, without needing to launch anything from terminal.

enduity, 2017-10-14:
So, I did a fresh raspbian lite install like you said, and installed it again. Still no bluetooth. I did find out how to pass my device on to it, and then it works, but it doesn't work itself. Still says it couldn't rewind.

morrolinux, 2017-10-14:
Good! We made one step forward.
1) are you aware the first connection it always fail (see known issues
section )
2) What is the output of tail -f /var/logs/Bluetooth_dev when you
connect/disconnect your phone?
Try to follow the log realtime and see what happens.
If new lines do not appear, it might be necessary to reload udev:
"systemctl force-reload udev systemd-udevd-control.socket
systemd-udevd-kernel.socket"

enduity, 2017-10-14:
1) Yes
2) It looks fine, but is remove and add supposed to be backwards? Also, analyzing the code, I don't get where the $ACTION variable is coming from. Granted, I just may be bad at bash.

morrolinux, 2017-10-14:
I'm not a big udev expert myself, but it seems to be some kind of
environment variable provided to the executed script by the specified
kernel rule... anyways it's just the variable that stores the add or remove
string needed for the bluetooth script to take action.
I think it's no problem with the order, maybe you first see "disconnected"
from the previous session and then "connected" from the current one.
have you tried running the command line for udev refresh? did it sort any
effect?

enduity, 2017-10-14:
I did the refresh, no change. It seems that it gets connected, and then immediately the remove call is executed. Sometimes my pc still shows Connected without music but sometimes it just Disconnects ("Paired"). It's just this on a loop:
```
pi@raspberrypi:~/mpradio-master/install $ sudo tail -f /var/log/bluetooth_dev
Executing bluetooth script...|add|
BT connected
BT bluez_source.18_5E_0F_19_ED_75.a2dp_source
Setting bluez_source to:  bluez_source.18_5E_0F_19_ED_75.a2dp_source
Executing bluetooth script...|remove|
Executing bluetooth script...|add|
BT connected
BT bluez_source.18_5E_0F_19_ED_75.a2dp_source
Setting bluez_source to:  bluez_source.18_5E_0F_19_ED_75.a2dp_source
Executing bluetooth script...|remove|
Executing bluetooth script...|add|
BT connected
BT bluez_source.18_5E_0F_19_ED_75.a2dp_source
Setting bluez_source to:  bluez_source.18_5E_0F_19_ED_75.a2dp_source
Executing bluetooth script...|remove|
Executing bluetooth script...|add|
BT connected
BT bluez_source.18_5E_0F_19_ED_75.a2dp_source
Setting bluez_source to:  bluez_source.18_5E_0F_19_ED_75.a2dp_source
Executing bluetooth script...|remove|
Executing bluetooth script...|add|
BT connected
BT bluez_source.18_5E_0F_19_ED_75.a2dp_source
Setting bluez_source to:  bluez_source.18_5E_0F_19_ED_75.a2dp_source
Executing bluetooth script...|remove|
Executing bluetooth script...|add|
BT connected
BT bluez_source.18_5E_0F_19_ED_75.a2dp_source
Setting bluez_source to:  bluez_source.18_5E_0F_19_ED_75.a2dp_source
Executing bluetooth script...|remove|
```

morrolinux, 2017-10-14:
I'm trying to reproduce this issue without luck. I guess if you are right,
and the remove action is launched too early when the device is stil
connected, that might be a good explaination of your pi's behaviour.
Can you debug this more precisely? (check all the variables content and
condition results by echoing them etc...) that'd be helpful.

enduity, 2017-10-15:
Bluetoothctl outputs this every connection:
<details>
  <summary>Click to expand</summary>
 
```
[CHG] Device 18:5E:0F:19:ED:75 Connected: yes
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 Class: 0x040400
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 Class: 0x000400
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 Class: 0x080400
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 Class: 0x0c0400
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001112-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller B8:27:EB:DB:92:86 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Device 18:5E:0F:19:ED:75 Connected: no
```
</details>
<br>
Nothing really on dmesg except for the bluetooth device being setup as input every connection:

```
[  916.173178] input: 18:5E:0F:19:ED:75 as /devices/virtual/input/input0
```

<br>

I really don't know where to look, since all the scripts seem to be working fine. What actually says if the device disconnects or disconnects it? The bluetoothctl output seems weird because the UUID seems to be tied to two Bluetooth LE devices connected to each other, but my computer doesn't have Bluetooth LE.

morrolinux, 2017-10-20:
Hi. sorry for the late response. What about just echoing the bluetooth script variables to stdin?

morrolinux, 2017-10-25:
pulseaudio was being killed upon mpradio restart, should be fixed now.

closed 2017-10-25 by morrolinux