morrolinux/mpradio #15: Trouble with RPi 3 Model B

matteocef, 2017-11-10:
Hey, loved the project. 
I'm trying all afternoon to try to make it works, but it doesn't. I'm not so used to linux, and i know very little. But i tried to have some logs and i receive this back:
 mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2017-11-10 18:14:29 UTC; 13min ago
 Main PID: 475 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─475 /home/pi/mpradio
           ├─681 pulseaudio -D
           ├─881 sh -c /usr/bin/sudo -u pi -s pulseaudio -D
           ├─882 /usr/bin/sudo -u pi -s pulseaudio -D
           ├─886 pulseaudio -D
           ├─887 pulseaudio -D
           └─888 pulseaudio -D

Nov 10 18:27:45 raspberrypi bash[475]: sox WARN formats: can't set sample rate 48000; using 44100
Nov 10 18:27:45 raspberrypi bash[475]: sox FAIL sox: `-' error writing output file: Broken pipe
Nov 10 18:27:45 raspberrypi bash[475]: sh: 1: /home/pi/PiFmRds/src/pi_fm_rds: not found
Nov 10 18:27:45 raspberrypi bash[475]: sox WARN formats: can't set sample rate 48000; using 44100
Nov 10 18:27:45 raspberrypi bash[475]: sox FAIL sox: `-' error writing output file: Broken pipe
Nov 10 18:27:45 raspberrypi bash[475]: sh: 1: /home/pi/PiFmRds/src/pi_fm_rds: not found
Nov 10 18:27:45 raspberrypi bash[475]: sox WARN formats: can't set sample rate 48000; using 44100
Nov 10 18:27:45 raspberrypi bash[475]: sox FAIL sox: `-' error writing output file: Broken pipe
Nov 10 18:27:45 raspberrypi sudo[882]:     root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c pulseaudio -D
Nov 10 18:27:45 raspberrypi sudo[882]: pam_unix(sudo:session): session opened for user pi by (uid=0)

and i cannot hear anything on the 88.8 freq. Can you help me? or i'm doing something wrong? :)

matteocef, 2017-11-10:
mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2017-11-10 18:32:07 UTC; 2s ago
 Main PID: 21666 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─21666 /home/pi/mpradio
           ├─21690 pulseaudio -D
           ├─21877 sh -c sox -t mp3 -v 1.3 -r 48000 -G "/pirateradio/Ghali - Sempre Me (Prod. Charlie Charles).mp3" -t wav - | /home/pi/PiFmRds/src/pi_fm_rds -ctl /home/pi/rds_ctl -ps "Ghali -
           └─21878 sox -t mp3 -v 1.3 -r 48000 -G /pirateradio/Ghali - Sempre Me (Prod. Charlie Charles).mp3 -t wav -

Nov 10 18:32:08 raspberrypi bash[21666]: sh: 1: /home/pi/PiFmRds/src/pi_fm_rds: not found
Nov 10 18:32:08 raspberrypi bash[21666]: sox WARN formats: can't set sample rate 48000; using 44100
Nov 10 18:32:08 raspberrypi bash[21666]: sox FAIL sox: `-' error writing output file: Broken pipe
Nov 10 18:32:08 raspberrypi sudo[21850]:     root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c pulseaudio -D
Nov 10 18:32:08 raspberrypi sudo[21850]: pam_unix(sudo:session): session opened for user pi by (uid=0)
Nov 10 18:32:08 raspberrypi pulseaudio[21856]: Daemon already running.
Nov 10 18:32:08 raspberrypi pulseaudio[21856]: pa_pid_file_create() failed.
Nov 10 18:32:08 raspberrypi bash[21666]: E: [pulseaudio] main.c: Daemon startup failed.
Nov 10 18:32:08 raspberrypi sudo[21850]: pam_unix(sudo:session): session closed for user pi
Nov 10 18:32:09 raspberrypi bash[21666]: sh: 1: /home/pi/PiFmRds/src/pi_fm_rds: not found

matteocef, 2017-11-10:
i also just tried reinstalling it and here the log:

pi@raspberrypi:~/mpradio-master/install $ sudo ./install.sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bluez-tools is already the newest version (0.2.0~20140808-5).
crudini is already the newest version (0.7-1).
libsox-fmt-mp3 is already the newest version (14.4.1-5).
pulseaudio-module-bluetooth is already the newest version (10.0-1+deb9u1).
python-dbus is already the newest version (1.2.4-1).
python-gobject is already the newest version (3.22.0-2).
python-gobject-2 is already the newest version (2.28.6-13).
sox is already the newest version (14.4.1-5).
bluez is already the newest version (5.43-2+rpi1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.11.0-3).
The following additional packages will be installed:
  libflac-dev libogg-dev libvorbis-dev
The following NEW packages will be installed:
  libflac-dev libogg-dev libsndfile1-dev libvorbis-dev
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,051 kB of archives.
After this operation, 3,680 kB of additional disk space will be used.
Err:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libogg-dev armhf 1.3.2-1
  Connection failed
Err:2 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libflac-dev armhf 1.3.2-1
  Could not connect to mirrordirector.raspbian.org:80 (93.93.128.193), connection timed out
Err:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libvorbis-dev armhf 1.3.5-4
  Unable to connect to mirrordirector.raspbian.org:http:
Err:4 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libsndfile1-dev armhf 1.0.27-3
  Unable to connect to mirrordirector.raspbian.org:http:
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/libo/libogg/libogg-dev_1.3.2-1_armhf.deb  Connection failed
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/f/flac/libflac-dev_1.3.2-1_armhf.deb  Could not connect to mirrordirector.raspbian.org:80 (93.93.128.193), connection timed out
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/libv/libvorbis/libvorbis-dev_1.3.5-4_armhf.deb  Unable to connect to mirrordirector.raspbian.org:http:
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/libs/libsndfile/libsndfile1-dev_1.0.27-3_armhf.deb  Unable to connect to mirrordirector.raspbian.org:http:
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
mkdir: cannot create directory ‘/pirateradio’: File exists
cp: cannot create regular file '/pirateradio/pirateradio.config': Read-only file system
mkdir: cannot create directory ‘/usr/lib/udev’: File exists
rm -f *.o
g++ -Wall	   -c -o mpradio.o mpradio.cc
g++ -Wall	   -c -o player.o player.cc
g++ -Wall	   -c -o fsrun.o fsrun.cc
g++ -Wall	   -c -o settings_provider.o settings_provider.cc
g++ -o mpradio mpradio.o player.o fsrun.o settings_provider.o
fatal: destination path 'PiFmRds' already exists and is not an empty directory.
rm -f *.o
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 rds.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 waveforms.c
gcc -Wall -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 pi_fm_rds.c
pi_fm_rds.c:105:21: fatal error: sndfile.h: No such file or directory
 #include <sndfile.h>
                     ^
compilation terminated.
Makefile:44: recipe for target 'pi_fm_rds.o' failed
make: *** [pi_fm_rds.o] Error 1
Installation completed! Rebooting in 10 seconds...

matteocef, 2017-11-11:
ok so, i tried reinstall everything from ZERO, and it downloaded everything and seams work just fine with usb stick and FM transmission: it just boot and start to transmit. But if i try to use the bluetooth i connect via bluetooth form my iphone, it change modality(cannot check the RDS) but if i start play something on the phone nothing listen!

morrolinux, 2018-01-22:
Please check it out on the latest version, I've fixed many issues :)

morrolinux, 2018-01-24:
Fixed.
Please reopen if you still have trouble related to this issue

closed 2018-01-24 by morrolinux

start-the-fire, 2018-01-24:
Hi,

I'm sorry to reopen the issue but I still have troubles get it working. Playing mp3's & changing the frequency vis usb thumb drive is working without any issues. My current problem is still the bluetooth playback. I can see & connect with my iPod to the rpi (the rpi is called now raspberrypi instead of mpradio it was before the last update?) without any problems - also the mp3 playback stops but when I try to play nothing happens. When I dosconnect the iPod it jumps back to mp3 playback.
With my smartphone I can see the rpi but everytime I try to connect via BT the connection fails :/

I also disabled the onboard wifi with "dtoverlay=pi3-disable-wifi" @ boot.config
One thing I will try out later is testing the code with another rpi3 - probably I grilled some parts of my rpi3 accidently :)

Some logs:
`pi@raspberrypi:~ $ sudo bluetoothctl
[NEW] Controller B8:27:EB:DF:19:64 raspberrypi [default]
[NEW] Device 78:00:9E:FC:2F:A5 Galaxy A3 (2016)
[NEW] Device 00:23:32:28:BC:A5 iPod
[CHG] Device 00:23:32:28:BC:A5 Connected: no
[CHG] Device 00:23:32:28:BC:A5 Connected: yes
[iPod]#

pi@raspberrypi:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Wed 2018-01-24 09:17:17 UTC; 1min 19s ago
  Process: 910 ExecStart=/bin/bash -l -c /home/pi/mpradio (code=killed, signal=TERM)
 Main PID: 910 (code=killed, signal=TERM)

Jan 24 09:17:17 raspberrypi bash[910]: Created low-pass FIR filter for audio channels, with cutoff at 12000.0 Hz
Jan 24 09:17:17 raspberrypi bash[910]: PI: 1234, PS: "02 Pastel.mp3".
Jan 24 09:17:17 raspberrypi bash[910]: RT: "02 Pastel.mp3"
Jan 24 09:17:17 raspberrypi bash[910]: Reading control commands on /home/pi/rds_ctl.
Jan 24 09:17:17 raspberrypi bash[910]: Starting to transmit on 107.0 MHz.
Jan 24 09:17:17 raspberrypi bash[910]: Terminating: cleanly deactivated the DMA engine and killed the carrier.
Jan 24 09:17:17 raspberrypi bash[910]: sox FAIL sox: `-' error writing output file: Broken pipe
Jan 24 09:17:17 raspberrypi bash[910]: sox WARN dither: dither clipped 257 samples; decrease volume?
Jan 24 09:17:17 raspberrypi bash[910]: sox WARN sox: `/pirateradio/02 Pastel.mp3' balancing clipped 2230 samples; decrease volume?
Jan 24 09:17:17 raspberrypi systemd[1]: Stopped Morrolinux Pirate Radio.

pi@raspberrypi:~ $ sudo systemctl status simple-agent
● simple-agent.service - simple-agent service
   Loaded: loaded (/etc/systemd/system/simple-agent.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-01-23 18:25:29 UTC; 14h ago
 Main PID: 522 (simple-agent)
   CGroup: /system.slice/simple-agent.service
           └─522 /usr/bin/python /bin/simple-agent

Jan 23 18:25:29 raspberrypi systemd[1]: Started simple-agent service.
Jan 23 18:25:29 raspberrypi bash[522]: mesg: ttyname failed: Inappropriate ioctl for device

pi@raspberrypi:~ $ sudo systemctl status bluealsa
● bluealsa.service - BluezALSA proxy
   Loaded: loaded (/lib/systemd/system/bluealsa.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-01-23 18:25:26 UTC; 14h ago
 Main PID: 454 (bluealsa)
   CGroup: /system.slice/bluealsa.service
           └─454 /usr/bin/bluealsa

Jan 23 18:25:26 raspberrypi systemd[1]: Started BluezALSA proxy.
pi@raspberrypi:~ $
`

morrolinux, 2018-01-24:
Yea please let me know if you can try on another Pi since i cannot
reproduce your issue :)

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

start-the-fire, 2018-01-27:
Hi,
I tried it today on my other rpi3 but with the same same result (I attached an video to show the problem)
I used my iPod, smartphone and pc as a bluetooth transmitter but everytime it ends with the same result
https://www.youtube.com/watch?v=ekbN9b0Wc_E

rpi3 running on latest stretch version and with built-in bluetooth - do you have experience with usb-bluetooth dongles?
cheers b

mich-de, 2018-01-27:
I've tryed with a Huawei smartphone in my car, no bt dongle

morrolinux, 2018-01-27:
@start-the-fire Unfortunately I don't have a Pi 3 to test on, do you have telegram? it would be better if we can chat realtime and maybe ssh to your Pi 
@mich-de Is there any issue?

mich-de, 2018-01-27:
Morro any issue, did you have some trick to extend the range of the radio? I need to put the radio close to the rpi zero w :)

start-the-fire, 2018-01-28:
@morrolinux yes - my number +43 `660 460 54 39`
I'm at the moment on the go but just send me text. I will setup remote ssh access in the meantime