morrolinux/mpradio #17: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11

agustinsabas, 2018-01-02:
Hi Morrolinux! I'm very interesting in your project! but i have this out:
(i tried without graphic entorn and its the same)

 sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Tue 2018-01-02 14:02:16 UTC; 3min 21s ago
  Process: 1249 ExecStart=/bin/bash -l -c /home/pi/mpradio (code=exited, status=0/SUCCESS)
 Main PID: 1249 (code=exited, status=0/SUCCESS)

Jan 02 14:02:15 raspberrypi bash[1249]: /etc/profile: line 4: id: command not found
Jan 02 14:02:15 raspberrypi bash[1249]: /etc/profile: line 4: [: : integer expression expected
Jan 02 14:02:16 raspberrypi bash[1249]: 88.5
Jan 02 14:02:16 raspberrypi bash[1249]: /pirateradio
Jan 02 14:02:16 raspberrypi bash[1249]: no bluetooth device provided, playing mp3s...
Jan 02 14:02:16 raspberrypi sudo[1275]:     root : TTY=unknown ; PWD=/ ; USER=pi ; COMMAND=/bin/bash -c pulseaudio -D
Jan 02 14:02:16 raspberrypi sudo[1275]: pam_unix(sudo:session): session opened for user pi by (uid=0)
Jan 02 14:02:16 raspberrypi pulseaudio[1281]: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jan 02 14:02:16 raspberrypi pulseaudio[1281]: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jan 02 14:02:16 raspberrypi sudo[1275]: pam_unix(sudo:session): session closed for user pi

morrolinux, 2018-01-22:
Please try with the latest pull, a lot of thing have changed since your trial :)

agustinsabas, 2018-01-24:
Ahora funciona perfectamente!

morrolinux, 2018-01-25:
great!

closed 2018-01-25 by morrolinux