morrolinux/mpradio #18: Not working on Pi Zero

donagrejja, 2018-01-20:
Hi.

I'm having some issues with this on my Pi Zero.

I've tried the 0.6 version both image download and the source download.
Here mpradio seems to start and work properly but when pairing it doesn't let me pair anything into it.
I can't find anything wrong in the config in the ready image.

I've also tried the git repo version and here it's also not working.
Pairing works properly here but it's missing options for playing sound to the device.
If i change the class to 0x240420 it appears as a car audio / multimedia unit but still doesn't present options for playing media to the devie.
My phone doesn't stay connected when i've connected to the device either.
mpradio is in status "dead" that is not run and pulseaudio is also not running and neither is the radio software.

Sorry to say but I have yet to find any version of this software that just works.

Any hints to what i'm doing wrong ? Or any information that you need to clarify things ?

Thanks.

morrolinux, 2018-01-21:
Hello
Due to lots of recent changes in pulseaudio's upstream, many things went
wrong with this code, which relied on few components which changed
interface (eg: bluez device name in PA's mixer, ect)
So I still have to update the code to fix all of them (which apparently
started to grow in number) and eventually publish a 0.7 ready-to-flash
image to make sure it works anytime.
As for the 0.6 image, it's tested to work with my bluetooth adapter, which
only support legacy pairing mode, but not with more recent adapters. That
problem should have been solved in the github's version as I later got a Pi
Zero and started developing there as well.
I'll update the code soon (scheduled this week), please stand by

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

donagrejja, 2018-01-21:
Thank you for your quick reply =)

donagrejja, 2018-01-22:
Pulled the latest version from git on a fresh raspbian stretch lite installation now and for me it still doesn't work.

I can pair my samsung galaxy s7 with the pi but when i do it doesn't show any media output options on the paired device.
Also the phone doesn't stay "connected" to the pi and mpradio doesn't spawn. seems like that since the script can't find any actual bluetooth device connected it doesn't spawn.

morrolinux, 2018-01-22:
I just committed the "big" update: if i didn't miss anything out, it should
work. i got rid of pulseaudio which was one of the main sources of troubles
to this project, and made some minor structural changes (yet to be
documented)

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

donagrejja, 2018-01-22:
I've tried the new commit with a freshly installed raspbian stretch lite.

Now it works to pair and see the audio options on the phone and also to play music.

However the sound is choppy so there might be some issues with settings in terms of performance ?

Thanks for the update =) Good job ! almost there ;)

morrolinux, 2018-01-22:
Ok great! are you working on the Pi Zero W?
I am testing it on that device and i see no trouble whatsoever: check the
CPU usage with top or htop, should be <= 70% (a great improvement in
performance since it was <= 95% with pulseaudio)

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

donagrejja, 2018-01-22:
Don't see any issue with the CPU usage actually now that I've chedked.

'
Tasks: 106 total,   1 running, 105 sleeping,   0 stopped,   0 zombie
%Cpu(s): 33.4 us, 15.8 sy,  0.0 ni, 50.2 id,  0.0 wa,  0.0 hi,  0.6 si,  0.0 st
KiB Mem :   444532 total,   308080 free,    31972 used,   104480 buff/cache
KiB Swap:   102396 total,   102396 free,        0 used.   361156 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
 1088 root      20   0    7404   2112   1840 S 24.4  0.5   0:34.04 pi_fm_rds
  367 root      20   0   51412   5096   4572 S  6.2  1.1   0:17.51 bluealsa
 1082 pi        20   0    4216   2600   2224 S  5.0  0.6   0:07.14 sox
  926 root      20   0       0      0      0 S  3.8  0.0   0:05.36 kworker/u2:1
 1090 root      20   0       0      0      0 S  3.4  0.0   0:03.49 kworker/u2:2
 1043 root      20   0    8212   3288   2724 R  2.2  0.7   0:03.25 top
 1081 pi        20   0   13900   2864   2532 S  0.6  0.6   0:00.98 arecord
  172 root      20   0       0      0      0 S  0.3  0.0   0:01.12 brcmf_wdog/mmc1
  362 root       0 -20       0      0      0 S  0.3  0.0   0:00.77 kworker/u3:2
  613 pi        20   0   11636   4176   3416 S  0.3  0.9   0:01.13 sshd
'

donagrejja, 2018-01-22:
Example

morrolinux, 2018-01-22:
From which source are you playing? have you tried playing from youtube?
maybe it's just about to give it a minute to settle down.

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

donagrejja, 2018-01-22:
I've tried with both mxplayer and youtube and it's the same.

But if you aren't experiencing these issues I don't know what it could be.

Possibly compability issues or such.

morrolinux, 2018-01-22:
Dunno, maybe try with your cellphone and let me know :)

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

donagrejja, 2018-01-22:
This was with my cellphone =) Samsung Galaxy S7

donagrejja, 2018-01-22:
Tested with another cellphone now (Nexus 5) and the result is still the same.

donagrejja, 2018-01-22:
Found the issue: https://github.com/raspberrypi/linux/issues/1402

The issue is with wifi and bluetooth enabled at the same time.

If i disable the built-in wifi it works =)

morrolinux, 2018-01-22:
> Found the issue: raspberrypi/linux#1402

are you on raspbian full desktop or minimal (just CLI)?

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

donagrejja, 2018-01-22:
the lite version. only CLI and no graphical interface.

donagrejja, 2018-01-22:
closed the issue now because this is no fault of your code =) thanks alot !

closed 2018-01-22 by donagrejja

morrolinux, 2018-01-22:
Adding to known issues, thanks for pointing out :)

donagrejja, 2018-01-22:
Just a hint on the install script.

'cp: cannot create regular file '/etc/pulse/daemon.conf': No such file or directory'

morrolinux, 2018-01-23:
That's right. I forgot to remove all references to pulseaudio which is no
longer being used.
Probably will come more clear on the next code cleanup :)

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

morrolinux, 2018-01-23:
> also i think the bt-setup service/scripts needs to be delayed or such.
>
> when i added "After: bluetooth.service" is could actually run at least.
> But was unable to pair to the device.
>
> '
> root@pi-radio:~# systemctl status bt-setup
> ● bt-setup.service - mpradio bluetooth init script
> Loaded: loaded (/etc/systemd/system/bt-setup.service; enabled; vendor
> preset: enabled)
> Active: failed (Result: exit-code) since Mon 2018-01-22 23:37:29 UTC; 3min
> 38s ago
> Process: 268 ExecStart=/bin/bash /bin/bt-setup.sh (code=exited,
> status=1/FAILURE)
> Main PID: 268 (code=exited, status=1/FAILURE)
>
> Jan 22 23:37:26 pi-radio systemd[1]: Started mpradio bluetooth init script.
> Jan 22 23:37:29 pi-radio bash[268]: Can't get device info: No such device
> Jan 22 23:37:29 pi-radio systemd[1]: bt-setup.service: Main process
> exited, code=exited, status=1/FAILURE
> Jan 22 23:37:29 pi-radio systemd[1]: bt-setup.service: Unit entered failed
> state.
> Jan 22 23:37:29 pi-radio systemd[1]: bt-setup.service: Failed with result
> 'exit-code'.
> '

Thanks for pointing out! you could make a pull request for the service file
if you wish.
simple-agent.service is responsible for the pairing, and I would need
further investigation, but to me it seems like it's the same issue as for
bt-setup (adding After=bluetooth.service)
let me know :)

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

donagrejja, 2018-01-23:
In what order should they start ? should simple-agent start first and then bt-setup ? or the other way around ?

morrolinux, 2018-01-23:
They don't really depend on each other, but they mutually depend on
bluetooth service.
I'm not home right now, therefore I cannot test. I suggest you to try
adding After=bluetooth.service in both services and see if it works

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

donagrejja, 2018-01-23:
I'm at work also =) I'll test later.

donagrejja, 2018-01-23:
There is still something fishy with the service startup.

It doesn't display as the hostname anymore it displays as "BCM43438A 37.4MHz Raspbe.."
Also it still doesn't want to pair. It just says unable to pair.

Bluetoothctl says "connected: yes" and then "connected: no" and the phone says unable to communicate with device/ unable to pair.

donagrejja, 2018-01-23:
For debugging purposes:

Controller reports this:
        Name: pi-radio
        Alias: pi-radio
        Class: 0x000000
        Powered: yes
        Discoverable: yes
        Pairable: yes
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d052B
        Discovering: no

morrolinux, 2018-01-23:
"Class: 0x000000" makes me think something something could be misplaced...
I don't know what's wrong, but none of this happened on my Pi Zero W, could
you try to git clone and reinstall the latest version (I updated this
afternoon) and see what happens?
Configuration files should come back in place as well.
After that, if you still have issues you could try do download and flash
the full dd image I provided on my website. that's to make sure we are
running on the exact same environment

-- 

*per il supporto tecnico rivolgersi a*

*Moreno Razzoli__________________________       Avvertenze ai sensi del
D.Lgs.196 del 30/06/2003 Le informazioni contenute in questo messaggio di
posta elettronica e/o files allegati, sono da considerarsi strettamente
riservati. Il loro utilizzo è consentito esclusivamente al destinatario del
messaggio, per le finalità  indicate nello stesso. Costituisce violazione
ai principi dettati dal D.Lgs. 196/2003: trattenere il messaggio stesso,
divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o
utilizzarlo per finalità  diverse. Qualora riceveste questo messaggio senza
esserne il destinatario Vi preghiamo cortesemente di darcene notizia via
e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo
dal Vostro sistema. Grazie.     *

donagrejja, 2018-01-23:
I actually just did a fresh install for debugging purposes. this is where i got these results.

The name displays properly when I add After: bt-setup.service to simple-agent.service.

I'll do some more tests.

donagrejja, 2018-01-23:
This change seems to have made it work. will test more.

/etc/systemd/system/bt-setup.service:
After=bluetooth.service simple-agent.service

/usr/lib/udev/bluetooth:
#!/bin/bash
# at each BT connection/disconnection start/stop the service mpradio
function log {
        sudo echo "[$(date)]: $*" >> /var/log/bluetooth_dev
}
BTMAC=${NAME//\"/}
if [ $ACTION = "remove" ]
then
        log "Stop Played Connection " $BTMAC
        sudo systemctl stop mpradio-bt@$BTMAC
        killall mpradio && killall sox
        sudo systemctl start mpradio
elif [ $ACTION = "add" ]
then
        log "Start Played Connection " $BTMAC
        sed -i "s/^defaults.bluealsa.device.*$/defaults.bluealsa.device \"$BTMAC\"/g" /home/pi/.asoundrc
        sudo systemctl stop mpradio
        killall mpradio && killall sox
        sudo systemctl start bt-setup
        sudo systemctl start mpradio-bt@$BTMAC
else
        log "Other action " $ACTION
fi

--

seems like the initialization done in bt-setup.sh is needed every time you start the connection.
like it was before defined in player.cc and then executed every time that mpradio was started.

but I have yet to confirm that it works 100%. can pair now though.

morrolinux, 2018-01-23:
Damn I forgot bt-setup brings the bluetooth interface UP so it's required to start before anything else relying on bluetooth! 
are you sure about bt-setup needing to be executed when a BT client connects? 
ATM its only functions are to:
1) bring the BT interface up and visible (which is already, if someone just connected to it)
2) force udev to detect new (BT) events and launch /usr/lib/udev/bluetooth accordingly (which already happened if this is executing) 
so I see no point in that.
Thanks for the advice on the bt-setup service thought, I'm updating the code right now :)