morrolinux/mpradio #23: status mpradio

mich-de, 2018-01-30:
Hi morro,
today i've re-installed your mpradio on RPI Wifi Zero. I think there is an error on /etc/profile
`**sudo systemctl status mpradio**`

```
mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Tue 2018-01-30 10:12:02 UTC; 2min 41s ago
  Process: 326 ExecStart=/bin/bash -l -c /home/pi/mpradio (code=exited, status=0/SUCCESS)
 Main PID: 326 (code=exited, status=0/SUCCESS)

Jan 30 10:11:54 raspberrypi systemd[1]: Started Morrolinux Pirate Radio.
Jan 30 10:11:59 raspberrypi bash[326]: /etc/profile: line 4: id: command not found
Jan 30 10:11:59 raspberrypi bash[326]: /etc/profile: line 4: [: : integer expression expec
Jan 30 10:11:59 raspberrypi bash[326]: mesg: ttyname failed: Inappropriate ioctl for devic
Jan 30 10:12:02 raspberrypi bash[326]: 107.0
Jan 30 10:12:02 raspberrypi bash[326]: /pirateradio
Jan 30 10:12:02 raspberrypi bash[326]: no bluetooth device provided, playing mp3s...
```

morrolinux, 2018-01-30:
Yup, I've yet to debug it, and it's strange since /etc/profile is not installed/modified by mpradio itself

mich-de, 2018-01-30:
Hi morro, i think i fixed by editing the /etc/profile
`#  PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games"
   PATH="/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games"`

mich-de, 2018-01-30:
pi@raspberrypi:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Tue 2018-01-30 11:06:53 UTC; 4min 2s ago
  Process: 825 ExecStart=/bin/bash -l -c /home/pi/mpradio (code=exited, status=0/SUCCESS)
 Main PID: 825 (code=exited, status=0/SUCCESS)

Jan 30 11:06:47 raspberrypi systemd[1]: Started Morrolinux Pirate Radio.
Jan 30 11:06:52 raspberrypi bash[825]: /etc/profile: line 4: id: command not found
Jan 30 11:06:52 raspberrypi bash[825]: /etc/profile: line 4: [: : integer expression expected
Jan 30 11:06:52 raspberrypi bash[825]: SSH is enabled and the default password for the 'pi' user has
Jan 30 11:06:52 raspberrypi bash[825]: This is a security risk - please login as the 'pi' user and t
Jan 30 11:06:52 raspberrypi bash[825]: mesg: ttyname failed: Inappropriate ioctl for device
Jan 30 11:06:53 raspberrypi bash[825]: 107.0
Jan 30 11:06:53 raspberrypi bash[825]: /pirateradio
Jan 30 11:06:53 raspberrypi bash[825]: no bluetooth device provided, playing mp3s...

pi@raspberrypi:~ $ sudo systemctl stop mpradio
pi@raspberrypi:~ $ sudo systemctl start mpradio
pi@raspberrypi:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-01-30 11:11:16 UTC; 3s ago
 Main PID: 996 ((bash))
   CGroup: /system.slice/mpradio.service
           └─996 (bash)

morrolinux, 2018-01-30:
so.. basically you removed the "sbin" directories from the /etc/profile? 
could you check for me the output of:
` $ which id ` and ` $ whereis id `

mich-de, 2018-01-30:
```
pi@raspberrypi:~ $ which id
/usr/bin/id
pi@raspberrypi:~ $ whereis id
id: /usr/bin/id /usr/share/man/man1/id.1.gz
pi@raspberrypi:~ $
```

mich-de, 2018-01-30:
`# /etc/profile: system-wide .profile file for the Bourne shell (sh(1))
# and Bourne compatible shells (bash(1), ksh(1), ash(1), ...).

if [ "`id -u`" -eq 0 ]; then
  PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"
else
   PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games"
fi
export PATH

if [ "${PS1-}" ]; then
  if [ "${BASH-}" ] && [ "$BASH" != "/bin/sh" ]; then
    # The file bash.bashrc already sets the default PS1.
    # PS1='\h:\w\$ '
    if [ -f /etc/bash.bashrc ]; then
      . /etc/bash.bashrc
    fi
  else
    if [ "`id -u`" -eq 0 ]; then
      PS1='# '
    else
      PS1='$ '
    fi
  fi
fi

if [ -d /etc/profile.d ]; then
  for i in /etc/profile.d/*.sh; do
    if [ -r $i ]; then
      . $i
    fi
  done
  unset i
fi`

morrolinux, 2018-01-30:
Why have you deleted the last comment?

mich-de, 2018-01-30:
simply because each time i reinstall the mpradio the /etc/profile return to default settings.

I solved the boot error by editing the /etc/profile as described here:
`https://askubuntu.com/questions/998503/bash-id-command-not-found`

mich-de, 2018-01-30:
```
pi@raspberrypi:~ $ sudo /bin/bash -l -c /home/pi/mpradio

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

107.0
/pirateradio
no bluetooth device provided, playing mp3s...
```

morrolinux, 2018-01-30:
At first sight, it doesn't seem like a mpradio problem to me, but rather a raspbian 9 configuration issue. Good to know there is a simple solution thought. 
as for the second issue: have you plugged an antenna on GPIO's pin 4? if you don't you won't have enough transmitting power to reach your FM receiver

mich-de, 2018-01-30:
strange Morro, i didn't touch nothing, https://github.com/morrolinux/mpradio/blob/master/install/pirateradio.config as you can see is your config file :)

morrolinux, 2018-01-30:
> strange Morro, i didn't touch nothing, https://github.com/morrolinux/
> mpradio/blob/master/install/pirateradio.config as you can see is your
> config file :)

That's it. Then the default freq. Is 107.0 so you should tune there! :)

morrolinux, 2018-01-30:
let me know if you solved :)

morrolinux, 2018-01-30:
PS: in /etc/profile you could also just edit the line:
 if [ "`id -u`" -eq 0 ]; then 
to 
 if [ "`/usr/bin/id -u`" -eq 0 ]; then 

This is happening because id (/usr/bin/id) is not in the PATH of the user executing /etc/profile upon login. shouldn't compromise mpradio's functionality anyways.

mich-de, 2018-01-30:
Morro solved, it is on 107.0 Hz no 88.8 :) regarding the /etc/profile didn't solved.

morrolinux, 2018-01-30:
Alright! 
the /etc/profile error is safe to ignore at the moment, as it doesn't interfere with mpradio's functionality and it's more a debian/user related issue rather than the coding itself.
Be aware that this will probably be gone in the future as I'm planning to launch the service with a different user, set different permissions, etc etc..

closed 2018-01-30 by morrolinux