morrolinux/mpradio #24: uninstall crash fstab external HDD

noppe91, 2018-02-01:
Hello,

i installed this on my Raspberry Pi 2 raspbian Jessie. And now my external Harddrive cant be mounted via fstab. Do you have any idea how i can solve this?

I already deleted the backup udev files.

thank you very much.

uname -a
Linux raspberrypi 4.9.78-v7+ #1084 SMP Thu Jan 25 18:05:49 GMT 2018 armv7l GNU/Linux

morrolinux, 2018-02-01:
Hi, could you check the content of /etc/fstab file?

noppe91, 2018-02-01:
Thank you for your fast reply. And no i mentioned i forgot some details:
fstab is not mount this USB Drive, but wenn i do sudo mount -a it is mounted.

/etc/fstab:
proc            /proc           proc    defaults          0       0                                                                                                            
/dev/mmcblk0p1  /boot           vfat    defaults          0       2                                                                                                            
/dev/mmcblk0p2  /               ext4    defaults,noatime  0       1                                                                                                            
UUID=d2027501-c0b2-4391-9c97-39b9bee0f941       /media/USB-HDD  ext4    defaults,nofail   0       0                                                                            
# a swapfile is not a swap partition, so no using swapon|off from here on, use  dphys-swapfile swap[on|off]  for that

morrolinux, 2018-02-01:
fstab should mount your drive automatically if it is connected at boot-time. 
Make sure you are not attaching the drive AFTER the boot, or it will be perfectly normal that you have to mount -a to mount it.

noppe91, 2018-02-01:
yeah that´s what it should do :) but since the installation it doesnt work anymore. 
dmesg with mount -a in cron works, and disk is found very early (sda): 
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.9.78-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1084 SMP Thu Jan 25 18:05:49 GMT 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 242688
[    0.000000] free_area_init_node: node 0, pgdat 80c6f440, node_mem_map ba3a1000
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @ba35c000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=dea
dline rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 940368K/970752K available (7168K kernel code, 486K rwdata, 2012K rodata, 1024K init, 770K bss, 22192K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
    lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c798fc   ( 487 kB)
       .bss : 0x80c7b000 - 0x80d3b9a4   ( 771 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]   Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000024] Switching to timer-based delay loop, resolution 52ns
[    0.000339] Console: colour dummy device 80x30
[    0.000361] console [tty1] enabled
[    0.000386] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000408] pid_max: default: 32768 minimum: 301
[    0.000780] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000795] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002084] Disabling memory control group subsystem
[    0.002194] CPU: Testing write buffer coherency: ok
[    0.002246] ftrace: allocating 22402 entries in 66 pages
[    0.053821] CPU0: update cpu_capacity 1024
[    0.053843] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.053889] Setting up static identity map for 0x100000 - 0x100034
[    0.056697] CPU1: update cpu_capacity 1024
[    0.056709] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.057735] CPU2: update cpu_capacity 1024
[    0.057744] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.058739] CPU3: update cpu_capacity 1024
[    0.058749] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.058879] Brought up 4 CPUs
[    0.058894] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.058899] CPU: All CPU(s) started in HYP mode.
[    0.058904] CPU: Virtualization extensions available.
[    0.060011] devtmpfs: initialized
[    0.073647] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.074046] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.074070] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.074849] pinctrl core: initialized pinctrl subsystem
[... 167 lines cut ...]
[    4.086283] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    4.401294] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    4.492803] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    4.591505] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    4.591764] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    4.664787] systemd-journald[141]: Received request to flush runtime journal from PID 1
[    6.107859] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    6.109389] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[... cut ...]
[   22.530944] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   22.648270] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   23.023597] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   23.828799] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   23.829064] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   25.358806] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   25.360482] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1

morrolinux, 2018-02-01:
Well that's weird: with both mpradio installed or removed, there's really nothing that should interfere with a simple fstab rule/entry. Nothing in the code / nor in the install/removal procedure and so on.. I have to say to me, il really looks like a generic Linux issue, take a look at some guy having the very same issue in similar (random) circumstances : https://superuser.com/questions/439966/why-is-my-usb-drive-not-mounted-during-startup
maybe try adding "auto" to the mount options?

morrolinux, 2018-02-06:
Seems like there's nothing left to say. Please reopen if needed

closed 2018-02-06 by morrolinux