morrolinux/mpradio #26: so WARN

evanstapler, 2018-03-06:
When I check the mpradio status, it says it is active in green but this is an error I get:

Mar 06 00:36:11 pi bash[1831]: sox WARN formats: can't set sample rate
Mar 06 00:36:11 pi bash[1831]: sox WARN wav: Length in output .wav head
lines 1-21/21 (END)

I can hear loud static every now and then, presumably when it switches songs. It appears to load from my USB dri

morrolinux, 2018-03-06:
Both warnings are safe to ignore and probably due (respectively) to the input file and the playback resume function (which can be disabled in settings file) But I'll have to check.
As for the static noise: your intuition is right, FM streaming stops while switching songs, hence the noise.
Another way of doing that is using a named pipe to an always active PiFmRds process, but it's tested to degrade the audio quality, so I opted not to do that. I still have found no better way of doing it.  
As for the Pulseaudio thing... I'm sorry, what do you mean by that? pulseaudio is no longer used in this project, are you running on the latest version?

evanstapler, 2018-03-06:
Thank you for the quick response! I'm a huge fan of this project. To clarify,

I spent a couple hours further diagnosing the problem last night. I have checked simple-agent, bluealsa, mpradio tstaus and they all appear to be functioning as intended. I have a pirateradio.config file on my USB flash drive pointing to 105.3 and btgain of 1.7 (default). I can actually hear my music, faintly, muddled by a lot of static and noise. But it's definitely there, and I can hear it over both bluetooth from my phone and just my USB flash drive. I don't know how to troubleshoot it from here, though.

I only have this problem with the latest version. The previous version I used could transmit with a USB drive, but I couldn't get the bluetooth to work.

As a side note, you may want to mention that mpradio uses RPi's GPIO 4 in the readme. Unless I missed it, I didn't find mention of it besides on your blog.

Thanks again, I'm looking forward to getting this working! I'll  be using this daily once it's up and running.

morrolinux, 2018-03-06:
Alright, so, maybe I'm missing something here. What problem are you exactly experiencing? the sound is muddled by lots of static noise? 
See, the problem here, is that mpradio uses PiFmRds to broadcast over FM thru the GPIO 4, and I'm not the author of PiFmRds (which is also why I haven't spent too much on documenting where to pin the antenna (which would be illegal to use in most countries and bla bla bla..) but I guess you're right I should at least mention that. ) 
So, back to your issue, that seems odd, but I'm not sure it's a software problem rather than an hardware related issue... As far as I know, it's best to use a 20-40cm wire on GPIO 4 (making sure about which pin is it, as there are 2 different notations for numbering GPIOs on the Pi) 
And transmitting on a "empty" (or likely empty) station would improve your signal strength as you wouldn't have to fight against the "official" FM transmitter station of your town, which usually delivers a much much stronger signal. 
As 105.3 seems (to me) a quite used and busy station, I think you should consider transmitting on some other frequencies that you find "free" on your radio (88.0 or something around 107 are usually good)
Please let me know :)

evanstapler, 2018-03-08:
Hello again! I have good news, I got it working!

I'm a CLI noob, so I'm still learning a lot. After installing mpradio on my RPi3 it worked without issue! So I reinstalled raspian on my Pi0w, and it still didn't work. So I reinstalled raspbian again, but this time I ran the following commands before installing mpradio:

`sudp apt-get update
sudo apt-get full-upgrade`

It ran for a couple hours. Then I reinstalled mpradio, and voila, it works like a charm! USB works, Bluetooth works (with WiFi disabled), and everything runs smoothly. I don't know if I ran into an isolated incident, but you may want to mention doing the update and full-upgrade commands before installing on a fresh image.

So my issue is solved! I don't know GitHub etiquette, should I close this issue now or give you a chance to respond? Can the owner of a repository close an issue? Or just the creator of an issue?

Thanks again for your project. It's is by-far the best implementation of RPi's Tx capabilities I've come across, and very easy to use. Next up, I'll be getting a NooElect SDR for more fun!

morrolinux, 2018-03-10:
That's great man!
Thanks for the feedback, I'll make sure everything is clear in
documentation :)
Ps: you can close the issue if you're able to. Otherwise I'll close it
myself

closed 2018-03-11 by morrolinux