morrolinux/mpradio #28: Booting Error

PJAlter, 2018-04-05:
Hello, I seem to be having a problem with the running of this program, I'm new to coding and Raspberry Pi so sorry if this is a really stupid problem, I posted the "sudo systemctl status mpradio" command results below.

● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-04-05 23:10:28 UTC; 3s ago
  Process: 25778 ExecStart=/bin/bash -l -c /home/pi/mpradio (code=exited, status=126)
 Main PID: 25778 (code=exited, status=126)

Apr 05 23:10:28 raspberrypi systemd[1]: Started Morrolinux Pirate Radio.
Apr 05 23:10:28 raspberrypi bash[25778]: /etc/profile: line 4: id: command not found
Apr 05 23:10:28 raspberrypi bash[25778]: /etc/profile: line 4: [: : integer expression expected
Apr 05 23:10:28 raspberrypi bash[25778]: mesg: ttyname failed: Inappropriate ioctl for device
Apr 05 23:10:28 raspberrypi bash[25778]: /bin/bash: /home/pi/mpradio: Is a directory
Apr 05 23:10:28 raspberrypi systemd[1]: mpradio.service: Main process exited, code=exited, status=126/n/a
Apr 05 23:10:28 raspberrypi systemd[1]: mpradio.service: Unit entered failed state.
Apr 05 23:10:28 raspberrypi systemd[1]: mpradio.service: Failed with result 'exit-code'.
pi@raspberrypi:~ $

morrolinux, 2018-04-06:
Hello
It seems you've cloned this repository with default name, please try  ' rm
-rf /home/pi/mpradio ' and follow the installation steps literally this
time.

Rationale:
If you clone this repo in your home, you'll have a folder named "mpradio"
in your home directory. But as of now, /home/mpradio should be the path of
the executable, which cannot be found since a folder with the same name is
already there.
Cloning this repo with any different name (mpradio-master as suggested in
the instillation steps) will avoid this conflict.

I have a pull request to fix this conflict and permit coning with default
name, but it's jet to be tested (quite busy these days) so for now try
following the installation instructions :)

Please let me know

PJAlter, 2018-04-06:
I am currently in school attempting the process again using Dataplicity to use the terminal while away from my home, I have just finished installing everything and everything seems to be working! YAY! When I get home I'll see if I am able to hear it on the radio and I'll let you know, but everything seems to be working fine, thank you so much for the code and the extra help can't wait to try it out! I've posted the status code below.

● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-04-06 11:47:53 UTC; 2min 41s ago
 Main PID: 663 (mpradio)
   CGroup: /system.slice/mpradio.service
           ├─663 /home/pi/mpradio
           ├─703 sh -c dd bs=15k skip=0 if="/pirateradio/Univeristy of Akron Hockey Warm up mix 2015-2016.mp3" | sox -t mp3 -v 1 -r 48000 -G  - 
           ├─704 dd bs=15k skip=0 if=/pirateradio/Univeristy of Akron Hockey Warm up mix 2015-2016.mp3
           ├─705 sox -t mp3 -v 1 -r 48000 -G - -t wav - compand 0.3,1 6:-70,-60,-20 -5 -90 0.2
           └─706 /home/pi/PiFmRds/src/pi_fm_rds -ctl /home/pi/rds_ctl -ps Univeristy of Akron Hockey Warm up mix 2015-2016.mp3 -rt Univeristy of

Apr 06 11:47:56 raspberrypi bash[663]: rdsUpdateInterval: 3
Apr 06 11:47:56 raspberrypi bash[663]: rdsCharsJump: 6
Apr 06 11:47:56 raspberrypi bash[663]: Output: fm
Apr 06 11:47:56 raspberrypi bash[663]: no bluetooth device provided, playing mp3s...
Apr 06 11:47:57 raspberrypi bash[663]: PLAY: /pirateradio/Univeristy of Akron Hockey Warm up mix 2015-2016.mp3
Apr 06 11:47:57 raspberrypi bash[663]: 0FILE SIZE: 15460884 DURATION: 966.386 BS: 15
Apr 06 11:47:57 raspberrypi bash[663]: seeking the track...
Apr 06 11:47:57 raspberrypi bash[663]: setting audio output to fm
Apr 06 11:47:58 raspberrypi bash[663]: sox WARN formats: can't set sample rate 48000; using 44100
Apr 06 11:47:58 raspberrypi bash[663]: sox WARN wav: Length in output .wav header will be wrong since can't seek to fix it

PJAlter, 2018-04-06:
It works! Yay! Thank you so much for your help!

closed 2018-04-07 by morrolinux