morrolinux/mpradio #30: Standardizing installation away from /home/pi

Hurricos, 2018-04-09:
Just a followup on what you said in response to the initial pull request:

> ... As of PiFmRds... yea it would make sense to move it to bin as well.. but maybe only if we can copy just the executable (which I haven't tried as well)

Ideally, as an end user, I would like to be able to `git clone` the repository, `cd mpradio && sudo make ./install.sh`, and then `sudo rm -rf ~/mpradio ~/PiFmRds` after installation and reboot while retaining full functionality. Really, as the end-user, I probably shouldn't be seeing anything I don't directly need to interact with during runtime in my `/home/pi`.

That's _ideally_, of course. It's hardly _necessary_ or even _useful_, but it would clean up things in the home directory, which is, I guess, a reasonable goal.

morrolinux, 2018-04-09:
That's right, if you can test moving PiFmRds binary to /bin/ (change references in code ect..) would be great. Otherwise I'll do test and do it myself in the next few days I think 👍

morrolinux, 2018-04-17:
I've just had a look at the code.
while it's technically possible move the pi_fm_rds executable to /bin/ without issues, there is a quite big reason not to do it:
* need2recompile.sh checks (at every boot) weather the Pi model has changed or not.
This is useful for people (like me) who tends to switch SD cards between raspberry(s): 
PiFmRds needs to be recompiled each time we change device, since its implementation heavily depend on the GPIO pinout (and some other stuff which is model-related and I don't know).
Therefore, the PiFmRds source folder would need to stay somewhere.
We could just copy the executable to /bin/ and not delete /home/pi/PiFmRds folder, leaving the end-user the choice to remove it (losing compatibility between devices) but I'm a bit afraid I'd be covered by issues from people who not know that. 
Or maybe we could just store the PiFmRds folder somewhere else out of the /home/pi directory... what do you think?

Hurricos, 2018-04-17:
There is a place for source code, actually ... in reality, /bin is _supposed to_ only be for binaries that need to be used very early on; the real place to put them would be /usr/bin. Now, for the sources which are NOT packaged with the distribution, /usr/local/src should be the right destination (as per [http://www.pathname.com/fhs/pub/fhs-2.3.html#PURPOSE18](the FHS), referenced in [https://unix.stackexchange.com/questions/10814/what-is-the-usr-local-src-folder-meant-for?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa](this question)).

This is more just a cleanup issue. It may be a good idea to wait until you / we have gone through the biggest parts of implementation before touching this -- we can always clean up at the end, after all, since it's just one big cleanup on the one script that does lots of stuff with flies, `install/install.sh`.

Hurricos, 2018-04-21:
I have completed what I believe is necessary. It could use some more testing, which I will do tomorrow, but feel free to pull in my branch and test it yourself.

Less changes are necessary than I suspected. Woohoo!

closed 2018-04-21 by Hurricos

morrolinux, 2018-04-21:
Thanks man, really appreciated