morrolinux/mpradio #36: Bluetooth streaming issues on RPI 3B+

Eva850, 2018-05-09:
Hello, I have a problem with bluetooth streaming. After connecting with a phone and when I start audio player, the sound is being transmitted for a few seconds and then it stops. I assume that PiFmRds stops working because there is no RF signal. When I disconnect bluetooth, it starts transmitting audio from pendrive, and it works fine without issues. When I connect with bluetooth again it works for a few seconds and crashes. What logs do you need? Is there anything else I need to check? I use it on Raspbian lite on Raspberry Pi 3b+

morrolinux, 2018-05-09:
Well, let's try with:
* your `pirateradio.config`
* `/var/log/bluetooth_dev ` from before connecting to when it disconnects 
* `systemctl status mpradio-bt@<tab here for bash completion on your BT mac address>`

Eva850, 2018-05-09:
I've probably explained it wrong, the bluetooth device stays connected and plays audio via bluetooth but after  few seconds RPi stops transmitting RF (107.0 MHz) but bluetooth remains connected. I assume it is not a bluetooth issue, but I'll post requested logs soon

Eva850, 2018-05-10:
Attaching requested logs:
[mpradio.log](https://github.com/morrolinux/mpradio/files/1991358/mpradio.log)
[bluetooth_dev.txt](https://github.com/morrolinux/mpradio/files/1991359/bluetooth_dev.txt)

morrolinux, 2018-05-10:
I'd really need logs from mpradio-bt *when* it's executing rather than when it's already dead. If you cannot catch it running, maybe try with `journalctl -f --unit=...`

Eva850, 2018-05-11:
Attaching another log. Unfrotunatelly I am unable to grab a log from mpradio-bt as it seems to be already dead when I try to grab a log immediatelly after bootup. The other thing is that in the moment that crash occurs, nothing new appears in mpradio log. I also attach screenshots from htop when BT streaming is fine and immediatelly after crash
[log.txt](https://github.com/morrolinux/mpradio/files/1994784/log.txt)

When playing over BT:
[image]
After crash:
[image]

morrolinux, 2018-05-11:
Does this happen also with other devices?
Have you disabled the Wi-Fi on the Pi3?

morrolinux, 2018-05-11:
Then please try disabling wifi first (see README - Known issues)

Eva850, 2018-05-11:
I've disabled the Wi-Fi but this issue still exists, is there anything else I can try?

morrolinux, 2018-05-11:
could be many things, I really would need to see the mpradio-bt logs. it's awkward that your logs are empty, here's what I see when I issue journalctl (with or without `-f`)

```
journalctl --unit=mpradio-bt@44\:C3\:46\:5B\:03\:DC.service
-- Logs begin at Thu 2016-11-03 17:16:44 UTC, end at Fri 2018-05-11 19:22:29 UTC. --
May 11 19:22:25 raspberrypi systemd[1]: Started mpradio bluetooth playback 44:C3:46:5B:03:DC.
May 11 19:22:25 raspberrypi mpradio[733]: freq: 88.0
May 11 19:22:25 raspberrypi mpradio[733]: btGain: 1.7 storageGain: 1
May 11 19:22:25 raspberrypi mpradio[733]: persistentPlaylist: 1
May 11 19:22:25 raspberrypi mpradio[733]: resumePlayback: 1
May 11 19:22:25 raspberrypi mpradio[733]: shuffle: 1
May 11 19:22:25 raspberrypi mpradio[733]: rdsUpdateInterval: 3
May 11 19:22:25 raspberrypi mpradio[733]: rdsCharsJump: 6
May 11 19:22:25 raspberrypi mpradio[733]: Output: fm
May 11 19:22:25 raspberrypi mpradio[733]: Format: mp3 -o -iname *.flac -o -iname *.wav
May 11 19:22:25 raspberrypi mpradio[733]: implementation: pi_fm_adv
May 11 19:22:25 raspberrypi mpradio[733]: Treble: 0
May 11 19:22:25 raspberrypi mpradio[733]: 44:C3:46:5B:03:DC
May 11 19:22:25 raspberrypi mpradio[733]: setting audio output to fm
May 11 19:22:26 raspberrypi mpradio[733]: sox WARN wav: Length in output .wav header will be wrong since can't seek to fix it
May 11 19:22:26 raspberrypi mpradio[733]: Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
May 11 19:22:26 raspberrypi sudo[740]:       pi : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/local/bin/pi_fm_adv --ps BLUETOOTH --rt A2DP BLUETOOTH --fr
May 11 19:22:26 raspberrypi sudo[740]: pam_unix(sudo:session): session opened for user root by (uid=0)

```
what else can you try... I couldn't really tell.. maybe just make sure you're running journalctl proprely, for example I see no escape backslashes in your command issues, have you used tab expansion?
PS: please prefer pastebin for pasting logs rather than sending attached txt(s)

Eva850, 2018-05-12:
Tried once more, the journalctl command syntax is like in your example with backslashes, but still no luck
https://pastebin.com/sg5jnjwE

morrolinux, 2018-05-12:
Well, that's because you're passing it your Pi's BT:AD:DR:ES:S$ which you shouldn't: the service is executed on the connected client's address!

Eva850, 2018-05-12:
Finally, I managed to get this log
https://pastebin.com/H6aYKf7p

morrolinux, 2018-05-12:
Ok great! now we should get something more comprehensible if you add `-f` option and start to capture before connecting the bluetooth, and keep logging until the RF signal gets lost

Eva850, 2018-05-12:
Here is the log
https://pastebin.com/c9cbt6Ae

morrolinux, 2018-05-14:
Hello, seems like everything starts fine but then something breaks during the playback.. could be bluetooth issues... 
could you post me the logs of bluealsa as well? 
Thanks

Eva850, 2018-05-14:
Maybe I've done something wrong but I got only this:
https://pastebin.com/V8spA5TU

morrolinux, 2018-05-14:
can you try with another BT adapter? an unstable one could be the cause of broken pipes

Eva850, 2018-05-15:
Unfortunately I don't have any USB Bluetooth adapter, I use the Raspberry pi's onboard one. But it works properly with OSMC, Bluetooth streaming works properly there

morrolinux, 2018-05-16:
could you try with the latest commit I've just pushed, and logging mpradio-bt just like you did last time?
I've just added more verbosity which I need to gather more data

Eva850, 2018-05-17:
I have updated using cd mpradio-master && git fetch origin && git reset --hard origin/master && cd install && sudo ./install.sh
This is the new log:
https://pastebin.com/dcc4zGDJ

morrolinux, 2018-05-17:
Awkward, that doesn't quite look like the output of the latest version.. could you post here the content of your .git/refs/heads/master inside mpradio-master?

Eva850, 2018-05-17:
Content of master:
b76cc0a20bd3c9d8762668b4511837bec398ca80

pi@raspberrypi:~/mpradio-master/.git/refs/heads $ git rev-parse HEAD
b76cc0a20bd3c9d8762668b4511837bec398ca80

I uninstalled, removed mpradio-master folder and re-installed but mpradio-bt log looks the same so I'm not posting it

Hurricos, 2018-05-18:
I cannot duplicate your results, @Eva850. I did the following:

- Wiped my SD card and restored a Raspbian Lite image from March 13/14 (see [this link](http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-03-14/) for sources) to it

- Mount the two partitions on the card, `boot` and `rootfs`, then `sudo touch boot/ssh` (I think that's the right one ... I just touched both `boot/ssh` and `rootfs/boot/ssh`). This enables ssh.

- Insert the microSD card into the Pi 3B+

- Connect Pi to >=2A power source and connect Pi with laptop via ethernet cable, configured for shared connection

- `ssh pi@IP_ADDR` (I use `nmap -sn <IP_RANGE_OF_ETH>/24` to find the address, which works if `sshd` is actually running on the Pi)

- On the Pi, `sudo apt update && sudo apt upgrade -y && sudo apt install git screen && reboot`

- Shell back in and `git clone https://github.com/morrolinux/mpradio` (this clones to `mpradio`, which works as well as `mpradio-master` since a recent [merge](https://github.com/morrolinux/mpradio/pull/33))

- `cd mpradio/install/ && sudo ./install.sh`

- Wait for reboot, then pair and run. I have a tuner that can pick up the default `107.0 FM`, but you may want to actually shell in to make sure things are running.

I suspect some step is being missed along the way - perhaps you forgot to `update` and `upgrade` before trying anything, or you're using a different image that comes with different defaults. In any case, let me know how it works.

Hurricos, 2018-05-18:
@morrolinux Following along with the above, it might be a good idea for me to open an issue and try some versioning compatibility, e.g. trying to run it on a fresh install of full-blown Raspbian, or (as with the above) on top of non-fresh installs. Just a thought.

Hurricos, 2018-05-18:
@Eva850 I highly suggest you try using the image I gave you from March 13/14. I have another guess: from your `htop` logs, I /think/ you're using a very recent version of Raspbian, which [apparently might have some issues with bluetooth](https://forums.gentoo.org/viewtopic-p-8220298.html).

Specifically, your `hciattach` is using a higher baud rate to the logic of /usr/bin/btuart, which some people are reporting causes instability.

morrolinux, 2018-05-20:
@Hurricos Good job of digging. Could be added to the "known isues" section when we'll know something for sure. 
@Eva850 keep us updated :)

morrolinux, 2018-05-29:
@Eva850 is the above solved? should I close this issue?

Hurricos, 2018-05-31:
@Eva850 I have a Raspberry Pi Zero W on the way -- it has the same updated bluetooth chipset, and so I should be able to reproduce your results this way and find a way around it when it arrives, likely by next week.

morrolinux, 2018-06-03:
UPDATE: The author of PiFmAdv just made an important change to the implementation, making it more stable and possibily fixing this issue as well. please update everything before you try again :)

Eva850, 2018-06-04:
Here is the log after updating and changing to pi_fm_adv. https://pastebin.com/amuMGij7
The issue is still there. But I must admit that I sometimes notice that during playback from pendrive song plays fine for a few seconds then transmission stops and the next song is being transmitted. I had this problem from the beginning but I haven't reported it, but now I think that the cause of both issues can be similar

morrolinux, 2018-06-04:
Seems like there was an issue with the latest update, please update *twice* and test again :)

Hurricos, 2018-06-04:
@morrolinux Do you mean the issue which occurs over Bluetooth, or the one which occurs via the pendrive?

morrolinux, 2018-06-04:
the pendrive one. But I think it's the same exact issue.
I'm currently running my Pi0w with mp3 and Bluetooth flawlessly

morrolinux, 2018-06-04:
@Hurricos Oh, I forgot: I'm running mpradio not on the latest raspbian (I just haven't updated it in a while - April 2018) that said, no issues so far with the latest mpradio.
could come handy to download this image and just update mpradio
https://www.mediafire.com/folder/qn62j675htott/mpradio

Eva850, 2018-06-06:
It seems I solved my issue. I use my RPI with OSMC so I have connected TSOP ir receiver to gpio18, it have been connected during all my tests. When I disconnected it, bluetooth works fine. Sorry for creating this unnecessary issue.

morrolinux, 2018-06-06:
Aahahah damn, that GPIO is reserved for the pushbutton, now I see why you got interrupted every 0.5 seconds.. that's the timing mpradio sets for checking the button state... 
Anyway, good to hear you solved your issue, don't forget to try the app (remote bluetooth) as well!

closed 2018-06-06 by morrolinux