morrolinux/mpradio #41: Cannot change radio frequency

branston, 2018-06-16:
First off all thank you for making this an easy install but I cannot for the life of me change the frequency at which the station broadcasted. I've tried the flash drive method and the app method neither of them work.

morrolinux, 2018-06-16:
Hello, I think I'd need the output of the following commands:
* `mount`
* `sudo fdisk -l`
* `cat /pirateradio/pirateradio.config`
* `ls -lah /pirateradio/pirateradio.config`
And please remember that the pendrive must be FAT32 partitioned on its first partition (sdX1)

As for the static: it might depend on which implementation are you using. PiFmAdv implementation should be the best of the best you can get in terms of quality and signal purity (have you tried yet?)

morrolinux, 2018-06-20:
Are you still there?

branston, 2018-06-20:
Yes, i will send it to you after I get off work today

branston, 2018-06-21:
So I restarted it and reflashed the rasbian image and reinstall your
packages and now it works fine. Let's me change frequency, the static is
gone. Works great!

eloverphx, 2018-06-28:
On the same topic, I'm using a pi zero without a pendrive, is there a way to change frequency on the fly without recompiling. I tried the app and it doesn't seem to be finding what it needs either.

loidor, 2018-06-28:
Or put the card in another computer and edit the file. That doesn't work in windows though.

eloverphx, 2018-06-28:
I didn't make it a read only unless its in the script. As far as the app, it seems to connect to the pi but when i try to change any of the settings nothing happens. The app says hold on then nothing. Maybe a permissions thing i might have to look it to maybe. Is there a log file i can look at for the app

morrolinux, 2018-06-28:
As far as I can tell, the app is working properly. It might be a filesystem permission issue or some sort of thing,  Plug a FAT32 USB drive if you have the chance, it should confirm my hypothesis.
Anyway I'll hook up the pi tonight or tomorrow and check myself what happens when no USB drive is present

morrolinux, 2018-06-28:
Well, I'm not sure, I've set up a Pi0W without pendrive and the app just succeed in changing settings on the fly.. BUT I'm not on the same version of the app as you are, so I think you should try the latest version of the app first.
I've updated the link in the readme.md

eloverphx, 2018-06-28:
Ok, I've updated the app and tried again and it didn't work. I was able to send a command for shutdown and restart from the app so that portion works great. I will burn a new image again and start a fresh run. I will use the newest raspbian minimal image and then use your script again and see what happens. I did look in my root directory and am not seeing the /mpradio folder. I just see /mpradio-ctl and I for got the other. I'll let you know my status soon.

morrolinux, 2018-06-29:
First off, is /pirateradio, not /mpradio you have to look for.
Second, the prefix "/" means the root folder, so you won't look for it in /home/pi/pirateradio but in /pirateradio

eloverphx, 2018-07-12:
hmm. I replied but it's not here, well long story short no luck still. I had some bad luck with my zero and the bluetooth failed to work anymore. I got another v1.3 zero and can't get it to connect using the same sd card so put those on the back burner. I hooked up my raspi 2 b+ and wrote a new card and ran the script to have the same issue. Still can't get the app to change the channels. I tried giving full r/w access to the pirateradio folder and all the files and that didn't work. I am using a Samsung s8+ not sure if there is something not allowing it to r/w to the pi over bluetooth, but it works when I use the command function. Any ideas?

morrolinux, 2018-07-12:
please post here your /pirateradio/pirateradio.config

eloverphx, 2018-07-12:
Ok well now the file is blank. I moved a copy of the file and looked and it had all the config info in there, then I connected my phone and tried the app then checked for any changes and it was empty again.

morrolinux, 2018-07-12:
1) The file you copied in place had comments or not?
2) On the app, after going to settings, did you see the parameters updating accordingly in a couple of seconds?

eloverphx, 2018-07-12:
I took the file straight from the /install folder and moved it to the root folder /pirateradio. There were no comments in the file and on the app I saw it show "bluetooth" instead of now playing for the first time then, I went to the setting page and it showed default channel of 88.8

morrolinux, 2018-07-12:
But the default frequency in the configuration file under the `install` folder is `107.0` so I guess the app is not able to get or save the config file locally. 
It could be many things... permissions, your download folder path different from the path indicated in the app, or even an OBEX FTP server  problem on the Pi. (which I doubt at this point)
But to sort few things out. 
1) Can you post the output of `systemctl status obexpushd`
2) Do you have Android studio or do you feel confident enough to install it and try to debug the app?
3) Otherwise I think more time will be needed, as I'm not in a good period with much free time, sorry about that

eloverphx, 2018-07-12:
`
● obexpushd.service - obex file transfer service
   Loaded: loaded (/etc/systemd/system/obexpushd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-07-12 21:01:07 UTC; 1min 59s ago
 Main PID: 515 (obexpushd)
   CGroup: /system.slice/obexpushd.service
           └─515 /usr/bin/obexpushd -B -n -o /pirateradio/ -s /bin/file_storage.sh -t FTP

Jul 12 21:01:48 raspberrypi obexpushd[515]: Name: pirateradio.config
Jul 12 21:01:49 raspberrypi obexpushd[515]: TAG: Name VALUE: pirateradio.config
Jul 12 21:01:49 raspberrypi obexpushd[515]: Type: binary
Jul 12 21:01:49 raspberrypi obexpushd[515]: TAG: Type VALUE: binary
Jul 12 21:01:49 raspberrypi obexpushd[515]: Path: .
Jul 12 21:01:49 raspberrypi obexpushd[515]: TAG: Path VALUE: .
Jul 12 21:01:49 raspberrypi obexpushd[515]: MODE: put
Jul 12 21:01:49 raspberrypi obexpushd[515]: script: storing file on disk...
Jul 12 21:01:49 raspberrypi obexpushd[515]: script: file saved to disk.
Jul 12 21:01:50 raspberrypi obexpushd[515}: script: done

I'll load it and see what I can see. 
Might be later but i'll let you know.

morrolinux, 2018-07-12:
As I suspected, the app is probably having issues storing the file, resulting in an empty file being sent back to the Pi when you push "apply settings"
But we should get more details about the app issues once you launch the app with your phone connected to Android Studio. 
Since the apk is a debug version, you should be able to see what happens realtime, by going to the "logcat" section down the window, after choosing the process "com.example.morro.telecomando" 
https://imgur.com/YQYyYko

morrolinux, 2018-08-01:
Closing for inactivity. Reopen if needed

closed 2018-08-01 by morrolinux