morrolinux/mpradio #44: `mpradio` broken by changes in bluez-alsa vers. 0.{8,9}

Hurricos, 2018-08-29:
Similar issue to what is seen [here](https://github.com/Arkq/bluez-alsa/issues/126).

Simply running `systemctl enable bluealsa` is not sufficient in `bluealsa` version 0.9, though it worked fine in version 0.7. This is because the `bluealsa` binary now requires you to specify on the command line which profile you would like to run with, if it is not `a2dp-source`.

From `/usr/bin/bluealsa --help`:

    By default only output profiles are enabled, which includes A2DP Source and
    HSP/HFP Audio Gateways. If one wants to enable other set of profiles, it is
    required to explicitly specify all of them using `-p NAME` options.

In order to override manually, we must override `systemd`. We start off with the following output for `systemctl cat bluealsa.service`:

    # /lib/systemd/system/bluealsa.service
    [Unit]
    Description=BluezALSA proxy
    Requires=bluetooth.service
    After=bluetooth.service

    [Service]
    Type=simple
    User=root
    ExecStart=/usr/bin/bluealsa

We must edit this and add lines to override `ExecStart`, via `sudo systemctl edit bluealsa.service`:

    [Service]
    ExecStart=
    ExecStart=/usr/bin/bluealsa -p a2dp-sink

Once we run `sudo systemctl daemon-reload && sudo systemctl restart bluealsa.service`, `bluealsa` and therefore `mpradio` will return to functioning normally.

I'm looking into what is required to automate the process of editing the `systemd` configuration for this; the issue I linked above discusses editing an init-file, but since `mpradio` leverages `systemd`, this is not a solution. Instead, we should follow [standard protocol for modifying `systemd` files](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/system_administrators_guide/sect-Managing_Services_with_systemd-Unit_Files#sect-Managing_Services_with_systemd-Unit_File_Modify).

As far as I can tell, this just then involves the addition of a few lines to `install.sh` near where we `systemctl enable bluealsa.service`:

    mkdir -p /etc/systemd/system/bluealsa.service.d/
    cat << 'EOF' >> /etc/systemd/system/bluealsa.service.d/override.conf
    [Service]
    ExecStart=
    ExecStart=/usr/bin/bluealsa -p a2dp-sink
    EOF

I will write the commit and test.

morrolinux, 2018-08-30:
Wow that seems annoying. Thanks for reaching out. I'll wait for your commit :)

Hurricos, 2018-09-04:
I've got it done, but it was a pain in the ass on account of the format of the installation file. Going to tackle a few things in that next, but will offer two separate pull requests.

Hurricos, 2018-09-04:
Pull request is up; I have done testing and it works fine. Feel free to close when you are satisfied with the results.

closed 2018-09-05 by morrolinux