morrolinux/mpradio #46: mpradio.service not starting at boot on a freshly-installed raspbian

morrolinux, 2018-10-02:
Leave your feedbacks here, if you happen to have this issue.
Right now I have no time to look into this, and more infos are needed.

ikoydg, 2018-10-07:
its random. i tried follow the instructions but it keeps happening. i run mpradio headless, i notice it when i cant find mpradio bluetooth. so i unplug and plug again the power source.

hopes this helps.

[image]

morrolinux, 2018-10-07:
I'm not too sure about that.. It's definitely different from what I've got, and seems more filesystem related... are you sure you're not running on a defective SD card?

ikoydg, 2018-10-07:
im sure its not defective. i install it on other sd card. still get this random error. 
sometimes it just reboots it self.

img 1.1 is not working for me.
but img 1.0 is fine.

ikoydg, 2018-10-08:
ohh, sorry.  its "mpradio.service not starting thread"
my img 1.0 is not running mpradio on start up. i waited 5 minutes still no songs playing. 
but when i connect the bluetooth and disconnect the bluetooth. mpradio start playing. 

Hope this helps

xpiii23, 2022-02-11:
Hello, 
I am trying to run mpradio on RPI Zero W 2 in Raspbian Bullseye Lite and having problem to start the service the actual error reads: 
mpradio.service - Morrolinux Pirate Radio
     Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2022-02-11 18:40:04 GMT; 6min ago
    Process: 584 ExecStart=/bin/bash -l -c sudo -E chrt -f 99 /home/pi/mpradio.py (code=exited, status=127)
   Main PID: 584 (code=exited, status=127)
        CPU: 124ms

Feb 11 18:39:59 mpradio systemd[1]: Started Morrolinux Pirate Radio.
Feb 11 18:40:04 mpradio bash[600]: /etc/profile: line 4: id: command not found
Feb 11 18:40:04 mpradio bash[584]: /etc/profile: line 4: [: : integer expression expected
Feb 11 18:40:04 mpradio systemd[1]: mpradio.service: Main process exited, code=exited, status=127/n/a
Feb 11 18:40:04 mpradio sudo[584]:       pi : PWD=/ ; USER=root ; COMMAND=/usr/bin/chrt -f 99 /home/pi/mpradio.py
Feb 11 18:40:04 mpradio bash[608]: chrt: failed to execute /home/pi/mpradio.py: No such file or directory
Feb 11 18:40:04 mpradio systemd[1]: mpradio.service: Failed with result 'exit-code'.
Feb 11 18:40:04 mpradio sudo[584]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Feb 11 18:40:04 mpradio sudo[584]: pam_unix(sudo:session): session closed for user root

Do you have any idea what is the problem ? 
Thnx