morrolinux/mpradio #47: mpradio bluetooth cant connect

ikoydg, 2018-10-04:
its working but, i want to broadcast songs via my phone bluetooth

after pairing and connecting my phone to the RPI 3, it disconnect and cant connect now.
MPRADIO is the name of the rpi bluetooth.
any solution here.

morrolinux, 2018-10-04:
Hi, is it a fresh install from the repo?
Could you try my latest full raspbian-mpradio image and see how does it compare? Does it work? Thanks!

morrolinux, 2018-10-04:
PS: when you use the SD card image, you need to wait for it to reboot after first boot after flashing. :)

ikoydg, 2018-10-04:
its already working, it transmit song and my radio can receive it.  works perfectly except for the BLUETOOTH part. 

on my phone side, already paired with MPRADIO bluetooth but cant connect.
on RPI 3 side, cant also connect, but it is paired already.

it can connect before i install MPRADIO, after installation of mp radio, fm transmitter works fine (auto start) but cant connect to bluetooth na.

what file do you need?

morrolinux, 2018-10-04:
Yeah, I was asking you to try my full system sdcard image to test out Bluetooth functionality. If you have a spare sdcard, please do. I really need to understand something, and that's the most efficient way for me to do it, with all the variables involved.
Anyway have you checked on the diagnostic commands I posted in the "troubleshooting" section of the README.md? Anything off?

ikoydg, 2018-10-04:
hmm, kinda weird, i install your img 1.1 version. it work. it transmit my music over my phone to rpi(MPRADIO) via bluetooth. thanks for this

but the APP is not working. "BT CONNECTION ERROR" (im connected,media)
cant change frequency via phone. hope you fix this. this is kinda great. or any fix for this?

for meantime will change config via ssh. 

BTW sometimes i just boot to "You are in emergency mode...............................please enter to continue" 
no buttons are working here. i just unplug ang replug the RPI. i run my RPI headless, i just notice this when i plug it into a monitor.

morrolinux, 2018-10-04:
The full image is a few commits behind this repo, so I figure there's an issue with something introduced lately. I'll have to figure out what.
As for the app: the frequency cannot be changed because the image you downloaded is configured in read only mode, therefore the settings file cannot be changed. You can solve this issue by simply attaching a FAT32 usb key with a copy of mpradio.conf in it.
Or just use "utility/roswitch.sh rw" to make it rw

morrolinux, 2018-10-04:
Just let me know if you succeed

ikoydg, 2018-10-05:
sorry for my bad english. i cant explain it properly. but i will try.

i attached the screenshot of the app. as you can see, its paired and connected to the bluetooth. but the app cant detect it. 
it do the "utility/roswitch.sh rw". 
my phone is connect to mpradio, connected for medua audio
i restarted several times. 

when i go to menu then select the SETTINGS, the app crashes.  

[image]

ikoydg, 2018-10-05:
hope this helps. i read the troubleshooting part. 
i also try,  
sudo systemctl start bt-setup
sudo systemctl enable bt-setup

its still INACTIVE

[image]

morrolinux, 2018-10-05:
What smartphone/Android version are you on?
Have you got the chance to try on another device?
As for the bt-setup service: It's fine for it to be inactive, It's a oneshot script that configures the interface and then quit

ikoydg, 2018-10-06:
tried on my 3 phones.
2pcs 0+ (O plus) ultra 3.0
1pc samsung galaxy note 4

still no luck. 

i also reburn/rewrite?? the image on sd card.
still no luck...

morrolinux, 2018-10-07:
just for testing purposes... when you're connnected on the CLI, try issuing 
`systemctl start rfcomm.service` 
and check if it's running with
`systemctl status rfcomm.service`
if it is, you can try pairing your phone again and the app should work.
Then we would have to figure out why it isn't starting automatically upon boot

ikoydg, 2018-10-09:
busy week :)
app still not working my friend.

morrolinux, 2018-10-09:
was the service running or not?

ikoydg, 2018-10-10:
i do status first, and the service is not running. 
then i start the service and check it again, and its active and running, tried again to pair other phones ( i borrowed my friend android phone, just for troubleshoot but still no work)

tried it on my phone still not working.

any logs? or something i can show?

thanks

morrolinux, 2018-10-10:
Did it fail at pairing? Or just with the app?

ikoydg, 2018-10-10:
successfully paired and playing audio via bluetooth. 
but the app is always crushing.

its ok for me na. for meantime i use ssh and edit pirateradio.config via ssh. 
its ok to closed for now.

thanks..

morrolinux, 2018-10-11:
I guess I'll have to investigate that further... keep you updated if I make progress

morrolinux, 2019-01-13:
Bluetooth audio should work fine
The app crash could also be an Android app problem because of storage path

morrolinux, 2019-01-25:
Have you tried bluetooth audio yet?