morrolinux/mpradio #48: not detecting usb or bluetooth devices, immediately terminates on launch

dedavlade, 2018-10-17:
pi@raspberrypi:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Wed 2018-10-17 19:46:13 CEST; 861ms ago
  Process: 4921 ExecStart=/bin/bash -l -c /bin/mpradio (code=killed, signal=ABRT)
 Main PID: 4921 (code=killed, signal=ABRT)

Oct 17 19:46:12 raspberrypi bash[4921]: Output: fm
Oct 17 19:46:12 raspberrypi bash[4921]: Format: mp3 -o -iname *.flac -o -iname *.wav
Oct 17 19:46:12 raspberrypi bash[4921]: implementation: pi_fm_rds
Oct 17 19:46:12 raspberrypi bash[4921]: Treble: 0
Oct 17 19:46:12 raspberrypi bash[4921]: no bluetooth device provided, playing mp3s...
Oct 17 19:46:13 raspberrypi bash[4921]: terminate called without an active exception
Oct 17 19:46:13 raspberrypi bash[4921]: CONTROL_PIPE_SETUP
Oct 17 19:46:13 raspberrypi systemd[1]: mpradio.service: Main process exited, code=killed, status=6/ABRT
Oct 17 19:46:13 raspberrypi systemd[1]: mpradio.service: Unit entered failed state.
Oct 17 19:46:13 raspberrypi systemd[1]: mpradio.service: Failed with result 'signal'.

*FAT32 USB connected with 1 .mp3 file and the config file, Bluetooth connected to my phone.*

dedavlade, 2018-10-29:
> Please provide more details and context

What else do you need?

нед, 28. окт 2018. у 19:24 morrolinux <notifications@github.com> је
написао/ла:

morrolinux, 2019-01-13:
Should be fixed with #58
Reopen if needed

closed 2019-01-13 by morrolinux