morrolinux/mpradio #49: Bluetooth pin problem

killertuxX, 2018-10-24:
Hello! My problem is when i try to connect to the pi my phone says that the pin code is wrong.
I used pin 0000 each try but it never connected.
However when i hooked the pi onto a monitor it started to work again. When i unplugged the monitor and boot the pi my phone didn't connect again... (Sorry for bad English). Could you please help me? I used your img file and a usb Bluetooth dongle.

morrolinux, 2018-10-25:
Hello, 
I see consistency is really hard to achieve when hardware is changing between devices.
To me, it looks like "bt-setup.service" and/or maybe "simple-agent.service" consequently, are being launched when the Bluetooth interface is not ready. They might be delayed by external factors (such as connecting a monitor) just enough for the bluetooth device to get ready, and therefore run fine then.

I would suggest you to try editing `/etc/systemd/system/bt-setup.service `  by adding 
`ExecStartPre=/bin/sleep 10` before the `ExecStart` directive. 
Reboot and see how it goes.

killertuxX, 2018-10-25:
Thanks! i changed 10 to 20 and everything started to work again! Thank you! You are very helpful!

closed 2018-10-25 by killertuxX