morrolinux/mpradio #50: Add option to disable BT and add storage directory

Crasha, 2018-10-26:
I'd like the option to disable bluetooth and add a storage path to where my music will live on the Pi instead of a USB stick.

I've tried running mpradio on my Pi 2 which doesn't have bluetooth and it fails;

`pi@raspberrypi:~ $ mpradio
freq: 88.30
btGain: 1.7 storageGain: 1
persistentPlaylist: 1
resumePlayback: 1
shuffle: 1
rdsUpdateInterval: 3
rdsCharsJump: 6
Output: fm
Format: mp3 -o -iname *.flac -o -iname *.wav
implementation: pi_fm_adv
Treble: 0
no bluetooth device provided, playing mp3s...
terminate called without an active exception
CONTROL_PIPE_SETUPAborted`

Any ideas?

morrolinux, 2018-10-27:
just add your songs in a folder called /pirateradio man :)

Crasha, 2018-10-27:
> just add your songs in a folder called /pirateradio man :)

Is that root? /pirateradio?

morrolinux, 2018-10-27:
yup

Crasha, 2018-10-27:
Working now, thanks! However, the "rdsPattern" seems to be ignored in the config. If I set it to "TEST" it won't show on the receiver as "TEST". Both the titles on the receiver show the "Artist name - Song name" - I would have thought one of these should change?

Crasha, 2018-10-27:
I'll have a go at implementing this and submit a PR.

Crasha, 2018-10-27:
The station name (PS) is also using the song name as well as the radio text (RT). I thought the station name could be configurable but it doesn't seem like it can. In the code, the station name is hard coded to "songName" for some reason. 

Could this be made configurable please?

Crasha, 2018-10-27:
I've added a fix for this in a branch. Could you give me permissions to make a pull request please?