morrolinux/mpradio #56: Bluetooth not piping to FM

stitcombe, 2019-01-05:
I'm having issues with my Pi Zero W when using bluetooth piping to FM. I've gotten the USB to play successfully over FM, so I know my setup works. I'm thinking it's an issue with the bt-setup script. Whenever I try to pipe from my phone over bluetooth the mpradio service fails.

I've tried stopping the bt-setup service, then restarting the mpradio service, confirmed it's running, then start the bt-setup service. As soon as I start piping over bluetooth, the mpradio service will fail again.

Here's the errors I'm getting from mpradio:

● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sat 2019-01-05 17:18:34 CST; 1min 32s ago
  Process: 16048 ExecStart=/bin/bash -l -c /bin/mpradio (code=killed, signal=ABRT)
 Main PID: 16048 (code=killed, signal=ABRT)

Jan 05 17:18:34 rPiZero bash[16048]: Output: fm
Jan 05 17:18:34 rPiZero bash[16048]: Format: mp3 -o -iname *.flac -o -iname *.wav
Jan 05 17:18:34 rPiZero bash[16048]: implementation: pi_fm_rds
Jan 05 17:18:34 rPiZero bash[16048]: Treble: 0
Jan 05 17:18:34 rPiZero bash[16048]: no bluetooth device provided, playing mp3s...
Jan 05 17:18:34 rPiZero bash[16048]: terminate called without an active exception
Jan 05 17:18:34 rPiZero bash[16048]: CONTROL_PIPE_SETUP
Jan 05 17:18:34 rPiZero systemd[1]: mpradio.service: Main process exited, code=killed, status=6/ABRT
Jan 05 17:18:34 rPiZero systemd[1]: mpradio.service: Unit entered failed state.
Jan 05 17:18:34 rPiZero systemd[1]: mpradio.service: Failed with result 'signal'.

Am I missing something here? Any ideas on how I can get this working?

Jesse27, 2019-01-06:
Hi I am also getting the same error, I haven't found a solution yet, I will post here if I find one.

morrolinux, 2019-01-06:
Hi.
Sorry ATM I don't really have much time for this. I'll give you some hints so that maybe you can figure it out.
The mpradio service doesn't really fail, it gets killed (by purpose) by the Bluetooth script (look at my schematic in the README) and mpradio-bt@.service gets called. So you have to look for those journals instead.
It worked before, but as mpradio heavily relays on external Linux services and utilities, if some implementation changes, it breaks.
That's why it is important to take a dd snapshot from time to time

stitcombe, 2019-01-11:
Thanks for the tips @morrolinux . 

The bluetooth script seems to be writing correctly to the log but I can't figure out why it's not starting mpradio-bt@.service correctly. When I check the mpradio-bt@.service status after connecting my bluetooth, it always says it's inactive. The script is killing mpradio, so I know it's running. Just doesn't seem to be starting mpradio-bt@. Any ideas?

morrolinux, 2019-01-12:
how do you check mpradio-bt@.service status?
I think you're doing it wrong

morrolinux, 2019-01-12:
have you tried with `journalctl --unit=mpradio-bt@yo:ur:bt:mac:add.service` ? because in systemd the `@` symbol means the unit expects a launch parameter, and the parameter is the bluetooth MAC address of the (cellphone?) you connect to it, so that's the actual unit you should check the logs for

stitcombe, 2019-01-12:
Sorry, still learning how to use linux.

I'm just getting a -- No entries -- when I run journalctl with my phone's BT address. I tried starting mpradio-bt manually and I'm still getting the --No entries--.

stitcombe, 2019-01-12:
I just found this is. Could this be causing the issue?

pi@PiZeroW:~ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-01-12 18:08:12 GMT; 1min 22s ago
     Docs: man:bluetoothd(8)
 Main PID: 441 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─441 /usr/lib/bluetooth/bluetoothd -C

Jan 12 18:08:12 PiZeroW systemd[1]: Started Bluetooth service.
Jan 12 18:08:12 PiZeroW bluetoothd[441]: Starting SDP server
Jan 12 18:08:13 PiZeroW bluetoothd[441]: Bluetooth management interface 1.14 initialized
Jan 12 18:08:13 PiZeroW bluetoothd[441]: Failed to obtain handles for "Service Changed" characteristic
Jan 12 18:08:13 PiZeroW bluetoothd[441]: Sap driver initialization failed.
Jan 12 18:08:13 PiZeroW bluetoothd[441]: sap-server: Operation not permitted (1)
Jan 12 18:08:13 PiZeroW bluetoothd[441]: Failed to set privacy: Rejected (0x0b)
Jan 12 18:08:13 PiZeroW bluetoothd[441]: Endpoint registered: sender=:1.4 path=/A2DP/SBC/Sink/1
Jan 12 18:09:23 PiZeroW bluetoothd[441]: Endpoint registered: sender=:1.4 path=/A2DP/SBC/Sink/2
Jan 12 18:09:28 PiZeroW bluetoothd[441]: /org/bluez/hci0/dev_40_98_AD_71_31_8C/fd0: fd(25) ready

morrolinux, 2019-01-13:
> Sorry, still learning how to use linux.
> 
> I'm just getting a -- No entries -- when I run journalctl with my phone's BT address. I tried starting mpradio-bt manually and I'm still getting the --No entries--.

Could you send me a screenshot where it's visibile which command have you issued to check the logs? just to disambiguate things a bit.

Not sure about the bluetooth.service log output.. I might be testing next week when I reach home. But I have to set up a Raspberry first, and debugging is quite time consuming

stitcombe, 2019-01-13:
I found the issue. I had to change 99-input.rules to get it to work. 

On mpradio install the file reads:
`ACTION==\"*\", SUBSYSTEM==\"bluetooth\", RUN+=\"/usr/lib/udev/bluetooth\"`

I changed it to:
`KERNEL=="input[0-9]*", RUN+=\"/usr/lib/udev/bluetooth\"`

I've confirmed this works with a fresh install of raspbian lite on my Zero W. One thing I did notice when I last ran the mpradio installer I saw the message "grep: /etc/udev/rules.d/99-input.rules: no such file or directory." No idea if that's causing an issue, but thought I provide that info.

morrolinux, 2019-01-13:
Good job! 
So, the rule is now more general (and could be triggered potentially by other kernel events too, such as connecting a usb dongle) but hopefully it won't cause issues since the Pi is supposed to be more or less "static" in term of connected peripherals.
You can create a PR if you want, or I can merge the change myself.
You let me know :)

stitcombe, 2019-01-13:
OK I've submitted PR. Brand new to this so hopefully I did it right. Cheers!

closed 2019-01-13 by stitcombe

Jesse27, 2019-01-15:
Hi thank you for your efforts however I am still having this issue even after the update any ideas regarding how to fix this.

morrolinux, 2019-01-15:
@Jesse27 what does `/etc/udev/rules.d/99-input.rules` file reads on your system?

Jesse27, 2019-01-21:
@stitcombe is it possible for you to upload your image with the latest version of mpradio, also do you have wifi enabled

stitcombe, 2019-01-22:
@Jesse27 sorry I no longer have access to that Zero W so I won't be able to create an image. I did have wifi enabled while testing but turned it off after everything was working properly, mpradio worked both ways for me.

Jesse27, 2019-01-22:
No worries, if you do have or gain access to another pi not necessarily a zero, it would be greatly appreciated if you could create an image. Additionally did your pi automatically connect to your phone or did you have to manually connect it each time.

stitcombe, 2019-01-23:
@Jesse27 I only had to pair once but manually connect the Bluetooth every time.