morrolinux/mpradio #59: Bluetooth still not piping to FM

Jesse27, 2019-01-15:
Hi, I am still having the same issue as #56 the update has not fixed it. I am trying to install on a clean install of both rasbian desktop and lite on a pi 3b+

Also is it possible for the pi to automatically connect to my phone on boot?

Thank you

morrolinux, 2019-01-15:
I'm unsure if this is 
* a different issue with the same effects
* the same issue as #56 (but not likely) 
* an update issue in which the bugfix in #56 haven't been applied when updating. 
So I would need more informations. I replied in #56 asking you to check something

Jesse27, 2019-01-15:
I'm at work at the moment but from memory it was:

KERNEL=="input[0-9]*", RUN+=\"/usr/lib/udev/bluetooth\"

I checked to make sure I Installed the new version. 

Also when I checked I don't think this file was installed 

/usr/lib/udev/bluetooth

So I copied it across from the install folder and looked through the install.sh to make sure nothing else was missing.

Additionally I installed mpradio on a clean image. 

I hope that helps a little.

morrolinux, 2019-01-15:
No need to rush, you can check all the above when you're home. 
The file `/usr/lib/udev/bluetooth` should have been already installed (as the installer script provides) or something went wrong during the installation

Jesse27, 2019-01-16:
Ok, I have just installed a fresh copy of raspian and mpradio.

The `/usr/lib/udev/bluetooth` file containes this code:
 `KERNEL==input[0-9]*, RUN+="/usr/lib/udev/bluetooth"`

And the `/usr/lib/udev/bluetooth` file installed correctly this time however it still does not work in bluetooth.

Here is the status of mpradio:
`pi@raspberrypi:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Wed 2019-01-16 18:34:48 AEDT; 8min ago
  Process: 408 ExecStart=/bin/bash -l -c /bin/mpradio (code=killed, signal=ABRT)
 Main PID: 408 (code=killed, signal=ABRT)

Jan 16 18:34:48 raspberrypi bash[408]: Output: fm
Jan 16 18:34:48 raspberrypi bash[408]: Format: mp3 -o -iname *.flac -o -iname *.wav
Jan 16 18:34:48 raspberrypi bash[408]: implementation: pi_fm_rds
Jan 16 18:34:48 raspberrypi bash[408]: Treble: 0
Jan 16 18:34:48 raspberrypi bash[408]: no bluetooth device provided, playing mp3s...
Jan 16 18:34:48 raspberrypi bash[408]: terminate called without an active exception
Jan 16 18:34:48 raspberrypi bash[408]: CONTROL_PIPE_SETUP
Jan 16 18:34:48 raspberrypi systemd[1]: mpradio.service: Main process exited, code=killed, status=6/ABRT
Jan 16 18:34:48 raspberrypi systemd[1]: mpradio.service: Unit entered failed state.
Jan 16 18:34:48 raspberrypi systemd[1]: mpradio.service: Failed with result 'signal'.
`

Jesse27, 2019-01-16:
Also it is worth noting the process number is not constant, I just started the service again and got the same error with this being the process `Process: 21143`

Jesse27, 2019-01-21:
@morrolinux Does the information I provided help in any way

morrolinux, 2019-01-22:
I'm afraid not, as it's perfectly normal for the `mpradio.service` unit to stop when a bluetooth device connects. Also, pid is generally dynamic at each program launch, not that it matters. 
There might be something useful in `/var/log/bt_dev` as the bluetooth script logs there. 
What happens on your phone? does it stay connected or does it get disconnected from the pi?

Jesse27, 2019-01-22:
So I just checked for the log and it wasn't there or anything related to Bluetooth. The phone stays connected to the pi even if the service stops.

Jesse27, 2019-01-25:
@morrolinux there is some interesting things in the system log about bulez and it not starting correctly, however it is worth noting that your 1.1 image does work. With that in mind I do have a solution that will save us time. My goal for this project is to be a headless Bluetooth handset set for my car which automatically connects to my phone (s8) when it is powered on. Is there any way you could implement that into a new image or if thats not possible would you be able to make me image of mpradio with wifi still enabled based on raspian desktop and lite so I could try to implement it myself. (I would rather build in desktop but I don't know how long the difference in start up time is). Also side note how difficult would it be to route call audio through the pi.

morrolinux, 2019-01-25:
> there is some interesting things in the system log about bulez and it not starting correctly

If the bluetooth device stays connected I'm not sure if the problem could be bluez. (see the scematic in the readme: simple-agent relies on bluez and allows you pairing) but why don't you post the incriminated log here then?

> however it is worth noting that your 1.1 image does work

what do you mean by that?

> Is there any way you could implement that into a new image or if thats not possible would you be able to make me image of mpradio with wifi still enabled based on raspian desktop and lite so I could try to implement it myself.

I haven't tried that, maybe it's a bluez service configuration setting or something like this. I would live this to you since in this period I'm really really busy and can't even afford to set-up another Pi and start working on it. I don't get why can't you just download the 1.1 image and modify that one, it could save you some time instead of installing raspbian lite and then mpradio on top of it (should be no big deal tho). If you succeed please let me know. We can always merge the changes, contributors are always welcome.

> (I would rather build in desktop but I don't know how long the difference in start up time is)

Really unnecessary and slower workflow to be connected directly on the pi for developing. you just need an IDE on your computer and a ssh/scp connection for the testing. You could also jut mount the home folder of your pi as a ssh filesystem and work directly there on your PC.
That's an advice of course.

> Also side note how difficult would it be to route call audio through the pi.

I'm not sure, it's managed by bluez profile(s) so most of the work should be already done, but I've never really tried to implement that. I was just not that enthusiast about making FM public phone calls ;-)

Jesse27, 2019-01-25:
Wow such a long comment, thank you for time. So I originally tried to modify your 1.1 image however I couldn't SSH into because it wouldn't connect to the network via wifi or lan on your particular image. Do you have any ideas?

morrolinux, 2019-01-25:
An HDMI adapter is very cheap to get, and you could configure everything with a USB keyboard attached via OTG. 
Otherwise there's a thing called "raspberry pi headless setup" in which you can configure such things like perhaps even the WIFI to connect to (so that you can reach it via ssh)

Jesse27, 2019-01-26:
Ok so I have started modifying the image, however the paired devices do not save on reboot, they unpair, is there any reason why this would happen.